adafruit/circuitpython #323: Feather M0 Adalogger does not have D8 or D7

ladyada, 2017-10-13:
D8 is the internal green LED
D7 is the interral Card Detect pin  (can be called SD_CD if ya like)
but those pins aren't exposed :)

tannewt, 2017-10-16:
Did you get D7 and D8 from `board`? It looks like you shouldn't https://github.com/adafruit/circuitpython/blob/master/atmel-samd/boards/feather_m0_adalogger/pins.c

ladyada, 2017-10-16:
yah it should have them, but doesnt yet :)
they're 'internal' to the adalogger - not exposed on the breakout. 
D7 is PA21
D8 is PA06

tannewt, 2017-10-16:
Ah ok! Is it urgent or can we have this a beginner task? It'd be to the file linked above:

- [ ] Add PA21 as `D7` and `SD_CD`
- [ ] Add PA06 as `D8` and `GREEN_LED`

ladyada, 2017-10-16:
its a good beginner task

mrmcwethy, 2017-11-12:
@Andon-A are you still working on this issue?

Andon-A, 2017-11-12:
@mrmcwethy Yes, I just haven't had a lot of spare time to actually build it and put it on the adalogger to make sure I didn't screw up.

Andon-A, 2017-11-27:
This is addressed in PR #461 - Marking as closed.

closed 2017-11-27 by Andon-A