adafruit/circuitpython #325: Consider using the circuitpython/lib/libm math library instead of -libm to save space

dhalbert, 2017-10-14:
The `atmel-samd` boards currently use `-libm`. This is supplied part of `glibc` in the `arm-none-eabi` toolchain. Various functions there such as `powf` and `sqrtf` end up using the double-precision float arithmetic routines (`__aeabi_dadd`, `dsub`, etc.), which occupy up a few thousand bytes.

There is an alternate simpler `libm` which is in our source tree which does not use doubles and is simpler in other ways. It's used for some other ports, including esp8266. I assume it might be numerically different as well, probably less precise.

When I tested switching `atmel-samd` to use the `libm` in our tree instead of `-libm`, the Gemma M0 firmware got smaller by 8780 bytes (!). This would give us the freedom to add a lot more stuff to the non-Express boards, and completely solves the current space squeeze (e.g., #322).

@tannewt @ladyada Do you have an opinion on this? Should we try to use the same `libm` across all `atmel-samd` boards (the size matters less for Express and M4).

The internal `libm` needs some work to remove compile warnings. Our `-Werror` errors out on all warnings.

ladyada, 2017-10-14:
!!!! lets try libm for gemma/trinket/non-exp for now since its the most useful!

dhalbert, 2017-10-15:
Gemma build using internal `libm`, complex arithmetic turned off, and `-finline-limit` disabled has 8312 bytes free.
Adding back complex uses 1576 bytes.
Re-enabling -finline-limit=50` (as an example) gains back about 1900 bytes, so there is headroom to use that again if necessary.

We could enable other stuff that might be useful even in small builds: #193 (`MICROPY_PY_ARRAY_SLICE_ASSIGN`, turned off in both builds), `reversed()`, etc. That's another discussion.

ladyada, 2017-10-15:
#1 most useful thing in small builds would be pwm/timer support.

dhalbert, 2017-10-15:
Turning on pulseio takes uses 7168 bytes, bringing free bytes down to 1092. Substantial, but it fits! I'll turn that on and add it to the pull request or make a separate one.

ladyada, 2017-10-15:
its worth it for sure, we can work around SLICE or reversed(), but if we dont have pwm support, we can't do servos, LED dimming, and a bunch of other things!

ladyada, 2017-10-15:
(if ya toss me a uf2 i will do a bunch of testing!)

dhalbert, 2017-10-15:
gemma and trinket uf2's attached in a zip
[pulseio-uf2-gemma-trinket.zip](https://github.com/adafruit/circuitpython/files/1385807/pulseio-uf2-gemma-trinket.zip)

dhalbert, 2017-10-16:
Limor [suggested](https://github.com/adafruit/circuitpython/issues/325#issuecomment-336605552) just the smaller builds, but happy to be consistent. Might be easier to support and test. I'll try to fix #331 before the merge anyway.

tannewt, 2017-10-16:
Thanks @dhalbert and @ladyada.

dhalbert, 2017-10-16:
Accomplished by #329.

closed 2017-10-16 by dhalbert