adafruit/circuitpython #326: 2 analog reads in a row get mixed up

ladyada, 2017-10-14:
on a cpx run:

```
import neopixel
from board import *
import analogio
import busio
from digitalio import *
import touchio
import time
import math
import adafruit_thermistor

light = analogio.AnalogIn(LIGHT)
speaker = analogio.AnalogOut(SPEAKER)
thermistor = adafruit_thermistor.Thermistor(TEMPERATURE, 10000, 10000, 25, 3950)

while True:
	    print("Temp: ", thermistor.temperature)
            print("light", light.value // 64)
```

but this works!
```
while True:
	    print("Temp: ", thermistor.temperature)
	    print("Temp: ", thermistor.temperature)
            print("light", light.value // 64)
            print("light", light.value // 64)
```

Adafruit CircuitPython 2.0.0-10-g922006dd5 on 2017-10-14; Adafruit CircuitPlayground Express with samd21g18

dhalbert, 2017-10-15:
I have a fix: swapping two lines. Pull request coming.

dhalbert, 2017-10-15:
Test prog:
```
import analogio
import adafruit_thermistor
from board import *
import microcontroller

light = analogio.AnalogIn(LIGHT)
thermistor = adafruit_thermistor.Thermistor(TEMPERATURE, 10000, 10000, 25, 3950)

N = 100
light_values = [None]*N
thermistor_values = [None]*N
cpu_temp_values = [None]*N

for i in range(N):
    light_values[i] = light.value
    thermistor_values[i] = thermistor.temperature
    cpu_temp_values[i] = microcontroller.cpu.temperature

print("light", light_values)
print("thermistor", thermistor_values)
print("temp", cpu_temp_values)
```

dhalbert, 2017-10-16:
Fixed by #327.

PR #327 (2017-10-15): Reinit ADC completely on each call to AnalogIn.value.
Fixes #326. AnalogIn used to keep around the ADE settings struct, and reference count it. However, now other code uses the ADC (e.g., microcontroller.core.temperaature), so the ADC needs to be completely re-inited each time it's used, since we don't know what state it's in. So now the`adc_config` and `adc_module` instances are just created each time on the stack.
closed 2017-10-16 by dhalbert