adafruit/circuitpython #328: Dcoument more differences between various builds

dhalbert, 2017-10-15:
The [Core Modules Support Matrix](https://github.com/adafruit/circuitpython/blob/master/shared-bindings/index.rst) lists some of modules are implemented in one port but not another. However, there are other differences that should be documented somewhere:
1. Other modules that are not in `shared-bindings` that differ, e.g. `framebuf` is not in non-Express samd builds.
2. Language features that differ, such as presence of absence of `reversed()`.

tannewt, 2018-05-22:
`framebuf` and other modules outside of `shared-bindings` aren't listed because they aren't supported. Ideally, only shared-bindings modules are supported because that structure ensures they are documented well too.

tannewt, 2018-05-22:
@dhalbert Mind doing this soon so we can close this issue? Thanks!

tannewt, 2018-05-24:
Done in #863

closed 2018-05-24 by tannewt