adafruit/circuitpython #335: Add better logic to turn individual shared-bindings features on and off.

dhalbert, 2017-10-16:
Need to coordinate across `main.c`, `Makefile`, and `mpconfigport.h`. Could add more stuff to `mpconfigboard.h` and/or `mpconfigboard.mk`. A single change should control the feature properly everywhere.

Motivated by https://github.com/adafruit/circuitpython/pull/333#discussion_r144948054.

sommersoft, 2018-09-17:
@deshipu brought this up in today's weekly meeting. Being on my phone and text only, I was a little late in getting some thoughts into the discussion.

I am going to need to implement a way to disable/enable the `pulseio.FrequencyIn` sub-module, since it leaves very little memory/flash left (or doesn't fit) on non-express SAMD boards.

My initial thought was to try and just `#ifdef` it in/out of shared-bindings with a `#define HAS_FREQ_IN` in `mpconfigboard.h`. I'm not even sure if that would work, or what sort of long define structure that approach would harvest. But, I think I've seen that as a non-desirable approach though a few PR reviews, anyway.

I don't necessarily want to tack this onto my pending PR, as I'm making some other changes already in `samd-peripherals`. Just want to be "ready" or at least discuss the direction that we want to go with this. I'm also fine with testing any suggested approaches with my use-case...

dhalbert, 2019-01-03:
Closed in favor of #1434, which is a duplicate but written up more clearly.

closed 2019-01-03 by dhalbert