adafruit/circuitpython #338: OneWire BusDevice

tannewt, 2017-10-16:
Factor out [this code](https://github.com/adafruit/Adafruit_CircuitPython_OneWire) into a OneWire BusDevice in [here](https://github.com/adafruit/Adafruit_CircuitPython_BusDevice). The original library can then be specific to the temp sensor.

caternuson, 2017-10-18:
New names also maybe?
`onewire_device.py` containing `OneWireDevice` class?
(matches the i2c and spi devices)

tannewt, 2017-10-18:
Yes please!

caternuson, 2017-11-15:
Work in progress lives here:
https://github.com/caternuson/Adafruit_CircuitPython_BusDevice/blob/master/adafruit_bus_device/onewire_device.py

caternuson, 2017-11-15:
Some questions:
- Since this just bit bangs a digital port, not sure what the set up and tear down should be for the context manager. Currently, there is nothing.
- This is a simple modification from the [previous code](https://github.com/adafruit/Adafruit_CircuitPython_OneWire) which wasn't really designed with the concept of representing a 'device'. Should the constructor also take in the device address? Then you wouldn't have to pass it in like is being done here:
https://github.com/adafruit/Adafruit_CircuitPython_OneWire/blob/master/main.py#L24

tannewt, 2017-11-16:
I think the context manager should manage one transaction. I believe in one wire that involves a scan to find the device first and nothing at the end.

It would be awesome if it represented one device. Anything not device specific can be top level functions.

I'd probably condense the read and write methods a bit to always operate on byte arrays.

caternuson, 2017-11-17:
The scan would be top level. Once that's done, you'd have the known addresses for what's on the bus and could then create specific devices using those addresses.

All that would be needed for a transaction is access to the digital pin.

tannewt, 2017-11-17:
Sounds good!

caternuson, 2017-11-17:
Where do you think these top level functions, like scan, should live?

Documenting some usages:
### Case 1
```python
>>> import board, busio
>>> import adafruit_ds18x20
>>> ow = busio.OneWire(board.D2)
>>> ds18 = ds18x20.DS18X20(ow)
>>> ds18.temperature
23.7
```
This would require the DS18X20 class to scan the bus and pick one (the first?) to use.
### Case 2:
```python
>>> import board, busio
>>> import adafruit_ds18x20
>>> ow = busio.OneWire(board.D2)
>>> ds18 = ds18x20.DS18X20(ow, address)
>>> ds18.temperature
23.7
```
A scan would not be required since the address is supplied. However, since the addresses are unique to every device, even of the same type, some form of a scan would be needed to determine the address.
### Case 3
```python
>>> import board, busio
>>> import adafruit_ds18x20
>>> ow = busio.OneWire(board.D2)
>>> outside = ds18x20.DS18X20(ow, address_outside)
>>> inside = ds18x20.DS18X20(ow, address_inside)
>>> outside.temperature, inside.temperature
-42.3, 23.7
```
Same comments as case 2.

tannewt, 2017-11-18:
Case 2 and 3. The coder will have to know the address somehow. Its pretty
easy for them to do a scan and select the first address if thats what they
want.

siddacious, 2017-11-21:
Wouldn't it also hold the 1-wire address class?

caternuson, 2017-11-21:
Probably, but I'm now questioning how necessary that is also. Could just keep it as simple as passing the 64bit bytearray around. Even though the bytearray is composed of [CRC][SERIAL NUMBER][FAMILY CODE], you end up using the entire 64bits as an address when communicating on the bus.

tannewt, 2017-11-27:
I'm reluctant to add scan to OneWire because then we'd need to know a bunch more about the protocol. Instead, I think a `OneWireBus` library could provide `scan` but also higher level transfer methods to read and write byte arrays.

I also like the idea of the OneWireAddress with two or three properties, serial number and family code.

caternuson, 2017-11-27:
My issue with the OneWireAddress class is that since CircuitPython doesn't support long integers, the 48 bit serial number will just be another byte array. And since the CRC and family code are only 1 byte, they can just be accessed easily and directly from the full byte array, ex: address[7] or address[0]. I think all a OneWireAddress class would do is provide a pretty syntax, ex: address.CRC vs. address[7].

But for now, assume we create a OneWireAddress class along with the OneWireBus library. Then example usage might look something like:
```python
>>> import board
>>> import adafruit_onewirebus
>>> import adafruit_ds18x20
>>> ow_bus = adafruit_onewirebus.OneWireBus(board.D2)
>>> ds18 = adafruit_ds18x20.DS18X20(ow_bus.scan()[0])
>>> ds18.temperature
23.7
```
where:
`adafruit_onewirebus`
  * imports `busio.onewire`
  * provides `scan()`, CRC math, read/write byte functions, `OneWireAddress` class, (more?)
  * `scan()` returns a tuple of `OneWireAddress` objects for each discovered device

`adafruit_ds18x20`
  * imports `adafruit_bus_device.onewire_device`
  * creates and uses `onewire_device` internally
  * will also need to import `adafruit_onewirebus` for access to `OneWireAddress` class

`adafruit_bus_device.onewire_device`
  * imports `adafruit_onewirebus` and provides context manager access

tannewt, 2017-11-27:
"Pretty syntax" is exactly why I think address is OK. Its way easier to discover `address.family` than `address[0]`.

Neither of the last two actually need to import `adafruit_onewirebus` because they can access the objects passed in even if they don't know about the type itself. I think you are thinking this because C expects to know the type of everything.

In Python, you can just pretend its a type you expect and as long as it provides the things you expect then it'll work. This flexibility makes it easier to reuse code. For example, we could make a "remote onewirebus" that another driver provides that would work just like the native onewirebus and the driver would just work. This is also useful for testing where you provide a fake bus that checks the underlying bytes.

caternuson, 2017-12-01:
Take a look at latest when you get a chance. I created a new repo for the working code:
https://github.com/caternuson/OneWire-Stuff
This will all probably live in different libraries when done but helps to have in one place for now. Copy all three files over and should be able to run the basic demo:
```python
>>> import board
>>> import adafruit_onewirebus
>>> import adafruit_ds18x20
>>> ow_bus = adafruit_onewirebus.OneWireBus(board.D2)
>>> ds18 = adafruit_ds18x20.DS18X20(ow_bus, ow_bus.scan()[0])
>>> ds18.temperature
23.7
```

tannewt, 2017-12-01:
As we talked about on [Discord](https://adafru.it/discord), it'd be good to:

- [ ] Remove _crc, _family_code and _serial_number from the address class in favor of reading it directly from rom in each property.
- [ ] Only expose write and readinto outside of onewirebus.

caternuson, 2017-12-06:
Done and somewhat done, scan() and reset() are still exposed. Need scan() to find devices. Need reset() to select rom.

Take a look. Same repo as above.

tannewt, 2017-12-06:
Looks really good! Want to cookiecutter it and PR it to https://github.com/adafruit/Adafruit_CircuitPython_OneWire? I think both bus and device can go there.

Can you make a new repo for the ds18x20 or do I need to?

caternuson, 2017-12-06:
Cool. Yah, I'll cookiecutter / PR the onewire stuff.

Please create a seed repo for the ds18x20 stuff. I'll PR into that.

tannewt, 2017-12-06:
Awesome! Done here: https://github.com/adafruit/Adafruit_CircuitPython_DS18x20

Beware that pylint will be a bit picky with your code. Install instructions are here: https://www.pylint.org/#install After its installed, its `pylint <python file>` to run.

tannewt, 2018-01-10:
@caternuson this done?

ladyada, 2018-01-15:
done and in bundle!!

closed 2018-01-15 by ladyada