adafruit/circuitpython #345: Examples are outdated

Andon-A, 2017-10-17:
The examples in the aptly-named /examples directory are out of date, some extremely so. It appears that they're all microPython based

These should be updated, replaced, or removed, and organized depending on what boards they run on.

My preferences:

- If it relies on specific hardware that there is no CircuitPython board comparable to, remove it.
- If its purpose is outside the scope of CircuitPython, remove it. 
- If we can adapt it to a specific CircuitPython board (I'm looking at you, Circuit Playground), do so.
- If it's an easy update, bring it into line with to-be-created guidelines.

tannewt, 2017-10-18:
@Andon-A this would be awesome! Are you up for it?

Andon-A, 2017-10-18:
Absolutely. I'm guessing most of them will end up being completely re-done or removed, but first up is the guidelines.

tannewt, 2017-10-18:
Guidelines look good to me!

fmorton, 2017-11-12:
I submitted a pull request for one converted example and added a new one in the process. Glad to do more if you would like.

tannewt, 2018-01-16:
At minimum we should delete unsupported docs and examples by 3.0.

fmorton, 2018-01-17:
I will get to this in the next couple weeks if no one else does.

Frank

tannewt, 2018-01-17:
Great! Thanks Frank!

commit 20dd3b1e (2018-02-22): Delete a bunch of docs, drivers and examples not relevant to CircuitPython.

This fixes #345 and fixes #215.
closed 2018-02-28 by dhalbert