adafruit/circuitpython #350: No easy way to reboot into safe mode on the Feather M0 Express

process1183, 2017-10-20:
While working on issue #286, I discovered that there is not an easy way (say shorting a pin to GND or 3V while pushing reset) to reboot the Feather M0 Express into safe mode. I messed up some test scripts and ended up getting the `CIRCUITPY` drive stuck in read only mode (read only from the PC) and had to temporarily flash firmware with safe mode forced on to fix it (this was easier than switching to Arduino and wiping the SPI flash).

The Feather M0 Express doesn't have a switchable safe mode implemented in `atmel-samd/boards/feather_m0_express/board.c`:

```c
bool board_requests_safe_mode(void) {
    return false;
}
```

If this is something that you would like done, I'd be happy to work on it.

tannewt, 2017-10-20:
Yes! That would be awesome. I think there may be a related bug about resetting from Python code too.

Are you on our [Discord](https://adafru.it/discord)? Its where we discuss things and have a weekly voice meeting.

process1183, 2017-10-20:
@tannewt Yep, I signed up for Discord. I may not be able to attend the weekly meetings though, because I'm at work all day.

tannewt, 2017-12-05:
I'm going to pick this up for 2.x. We want the ability to reset to the bootloader so I'll cover this at the same time.

Adding the ability to reset to the bootloader next reset press means we can provide the same out-of-box experience to the CircuitPlayground Express even though it'll ship with CircuitPython going forwards. We'll ship with demo code that make subsequent resets go to the bootloader rather than requiring a double tap. The double tap behavior will happen once that demo code is replaced though.

tannewt, 2017-12-06:
This is fixed in 2.x and will be merged into master soon.

PR #474 (2017-12-06): Introduce reset mechanics to microcontroller.
This allows one to configure how a subsequent reset will behave and
also trigger a reset.

Fixes #350 and fixes #173
commit b4d36990 (2017-12-06): Introduce reset mechanics to microcontroller.

This allows one to configure how a subsequent reset will behave and
also trigger a reset.

Fixes #350 and fixes #173
closed 2017-12-06 by tannewt