adafruit/circuitpython #358: i2c read_into hardfaults if given bytearray of size zero

ladyada, 2017-10-22:
yay safemode, here's a code snippet. no i should not be trying to read 0 bytes. but it also should not hardfault :D

Adafruit CircuitPython 2.1.0 on 2017-10-17; Adafruit Metro M0 Express with samd21g18

```

    self.sgp_run_profile(["iaq_init", 0x2003, 0, 0.01])

    def sgp_run_profile(self, profile):
        name, command, signals, delay = profile
        print("running profile: %s, command 0x%x, %d, delay %f" % (name, command, signals, delay))
        with self._device:
            self._device.write(bytes([(command >> 8) & 0xFF,
                                      command & 0xFF]))
            time.sleep(delay)
            crc_result = bytearray(signals)
            self._device.read_into(crc_result)
            print("\tRaw Read: ", crc_result)
```

tannewt, 2017-10-26:
This is fixed in 2.x.

jerryneedell, 2017-10-27:
In 2.x this breaks i2c_device.py in bus_device 
in line 56 it sends a zero length write"
```def __init__(self, i2c, device_address):
        # Verify that a deivce with that address exists.
        while not i2c.try_lock():
            pass
        try:
            i2c.writeto(device_address, b'')
        except OSError:
            raise ValueError("No I2C device at address: %x" % device_address)
        finally:
            i2c.unlock()

        self.i2c = i2c
        self.device_address = device_address
```

breaks a lot of I2C drivers tah call i2c_device!

dhalbert, 2017-10-27:
see also https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/issues/9

jerryneedell, 2017-10-27:
If there is another acceptable way to test the aliveness of the device, I'll be happy to implement it in i2c_device.py.

dhalbert, 2017-10-27:
Scott and I discussed whether to fix the original bug in a different way besides excluding buffers of size zero. I will look at it.

jerryneedell, 2017-10-27:
ok I tried just doing a read from it and that seems to work.
```junk=bytearray(1)
i2c.readfrom_into(0x5a,junk)
```

works  with a device at 0x5a
throws OSError 19 with an invalid address

jerryneedell, 2017-10-27:
I tried the following change in i2c_device.py and it works OK  with a simple test with a mpr121 installed and an si7021 not inttalled. If this is "i2c safe" it's a simple change.
```jerryneedell@Ubuntu-Macmini:~/projects/test$ diff i2c_device.py ../adafruit_github/Adafruit_CircuitPython_Bundle/libraries/helpers/bus_device/adafruit_bus_device/i2c_device.py 
56,57c56
<             bytesread=bytearray(1)
<             i2c.readfrom_into(device_address, bytesread)
---
>             i2c.writeto(device_address, b'')
```

jerryneedell, 2017-10-28:
This also break rgb.py in adafruit_rgb_display:
```Adafruit CircuitPython 2.1.0-3-gbd7abcd on 2017-10-28; Hacked Feather M0 Express with 8Mbyte SPI flash with samd21g18
>>> import tft_m0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "tft_m0.py", line 32, in <module>
  File "libraries/drivers/rgb-display/adafruit_rgb_display/rgb.py", line 93, in fill
  File "libraries/drivers/rgb-display/adafruit_rgb_display/rgb.py", line 82, in fill_rectangle
  File "libraries/drivers/rgb-display/adafruit_rgb_display/rgb.py", line 54, in _block
  File "libraries/drivers/rgb-display/adafruit_rgb_display/rgb.py", line 131, in _write
  File "libraries/drivers/rgb-display/adafruit_rgb_display/rgb.py", line 131, in _write
ValueError: Buffer must be at least length 1
>>> 
```
This is an SPI device and the problem comes from this line 82:
```
      self._block(x, y, x + w - 1, y + h - 1, b'')
```
which passes an empty byte array.
I have not found a simple fix to this yet.

dhalbert, 2017-10-30:
Testing this to see if I can fix it another way. I'm using an SHT31 humidity sensor at I2C 0x44 (no library, but it doesn't matter).
```
import board,busio
i2c = busio.I2C(board.SCL, board.SDA)
i2c.try_lock()
b0 = bytearray(0)
i2c.readfrom_into(0x44, b0)
```
hard-crashes only the first time the device is referenced. If I press reset without power-cycling, I get an `OSError` the second time. I have to power-cycle the SHT31 to cause a hard crash.

dhalbert, 2017-10-30:
Just documenting base error: ASF3 does not handle an i2c read request of length zero. CPy/Mpy passes in a zero length bytearray as a buffer starting at 0x0, with length 0. The read request eventually gets passed into `asf/sam0/drivers/serceom/i2c/i2c_sam0/i2c_master.c`, routine `_i2c_master_read_packet`. There is some handling of a zero length request, but critically at the end of the `while` read loop, there is this code:
```
		/* Save last data to buffer. */
		_i2c_master_wait_for_sync(module);
		packet->data[counter] = i2c_module->DATA.reg;
```
So it tries to write a single byte to `packet->data[counter]` even when the original length is zero. `data` is `0x0`, `counter` is 0 here.

Easiest fix at the Cpy level is to forbid i2c reads of length zero, but not writes.

tannewt, 2017-10-31:
This should actually be fixed now by f738996164fc270015fdb53ee38b24f7823cbbd8. Thanks @dhalbert !

dhalbert, 2017-11-03:
As discovered by @jerryneedell and @deshipu, https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display expects to be able to pass zero-length SPI buffers (and have nothing happen). Perhaps we should just handle this edge case and do nothing. The Arduino SPI library will take a zero-length buffer and do absolutely nothing.

Forbidding zero-length SPI reads may still make sense. As noted above, zero-length I2C reads exercise a bug, and don't have a use case, I think.

dhalbert, 2017-11-07:
Fixed by #404. Thanks @deshipu

PR #370 (2017-10-25): shared-bindings: Check that I2C and SPI reads and writes are given a …
…buffer of at least 1.

Fixes #358
commit bd7abcda (2017-10-25): shared-bindings: Check that I2C and SPI reads and writes are given a buffer of at least 1. (#370)

Fixes #358
closed 2017-11-07 by dhalbert