adafruit/circuitpython #360: Aggregate pulse states between all CircuitPython repos.

tannewt, 2017-10-23:
GitHub pulse is awesome in pulling stats for an individual in the last week. It would be awesome to have a python script that can run in the [Adafruit bundle](https://github.com/adafruit/Adafruit_CircuitPython_Bundle) and [Community bundle](https://github.com/adafruit/CircuitPython_Community_Bundle) and aggregate the stats for all of the repos.

Start with the stats on the [Pulse page](https://github.com/adafruit/circuitpython/pulse). Later we can add other things like new contributors.

hukuzatuna, 2017-10-25:
To be clear, @tannewt , this is a request for a standalone Python program that you would run, for example, on your laptop, which would interact with the GitHub API to extract and aggregate the desired stats? I am stating it this way because I don't see a way to do this using Pulse within GitHub itself. Perhaps I'm just not familiar enough with it, though.

tannewt, 2017-10-25:
Yup, python script or js webpage (so everyone can see) would be perfect.

tannewt, 2018-01-16:
This is done in https://github.com/adafruit/adabot/commit/716b85d4d6ca1a8087eece88189e72a6c877ce21

closed 2018-01-16 by tannewt