adafruit/circuitpython #361: Add pycodestyle checking to Travis

tannewt, 2017-10-23:
Add [pycodestyle](https://github.com/PyCQA/pycodestyle) checking to Travis builds for all CircuitPython drivers and examples.

[autopep8](https://github.com/hhatto/autopep8) can be used to clean up failing code and get travis back green.

willingc, 2017-10-23:
Hi @tannewt,

Just to clarify the use of `autopep8`, since it's not entirely clear from above. I think it's fine to have a tool to check the style of PRs. I would be cautious about using the tool to automatically change code as that can introduce errors and unexpected results.

Just an fyi, well maintained tools can be found in the PyCQA org: 
https://github.com/PyCQA

tannewt, 2017-10-23:
@willingc I don't intend on automatically running autopep8. I was thinking it'd be used by someone to fix the backlog of existing files we have. That way when Travis style checking is green when enabled.

Thanks for the links to those tools! Add love to be able to add checks for when certain Python funcitonality is used as well. For example, I'd love to keep bit manipulation (`|` and `&`) out of examples and only in driver code.

willingc, 2017-10-23:
@tannewt Cool. We're in agreement. It wasn't clear from the earlier message :-) 

One thing that we have done on other projects is temporarily activate the pep8 style checking and then let it pass even with errors until the backlog has been worked through.

tannewt, 2017-10-23:
Agreement is good! Thanks for the sanity check @willingc

tannewt, 2018-01-16:
This is superseded by https://github.com/adafruit/circuitpython/issues/475 which is almost done.

closed 2018-01-16 by tannewt