adafruit/circuitpython #364: I2C & SPI BusDevice readinto/read_into

ladyada, 2017-10-24:
hiya it appears i2c's busdevice uses read_into but spi uses readinto
maybe lets standardize on one?

tannewt, 2017-10-25:
Yes!

mrmcwethy, 2017-10-30:
i would happy to do this one.  "read_into" is mo' python-like, no?

deshipu, 2017-10-31:
That is going to break a lot of drivers.

ladyada, 2017-10-31:
@deshipu we would leave in a wrapper

tannewt, 2017-10-31:
@mrmcwethy looks like `readinto` is already used in CPython: https://docs.python.org/3/search.html?q=readinto Thanks for taking this on!

I think i2c is `read_into` because it was originally `readfrom_into` and then I changed how the address was managed.

As @deshipu and @ladyada point out, please start out by leaving `read_into` with a warning that its deprecated. Then, if you like, go through and update drivers. We can remove it once they are all fixed.

mrmcwethy, 2017-11-01:
I will place recreate a new method in i2c from readinto, which will be functionally equivalent to i2c.read_into.  I will also change the docs to say the i2c.read_into is obsolete and will be deprecated.

Question: what version of circuitpython will i2c.read_into() be removed?

mrmcwethy, 2017-11-01:
Also I assume that this issue only effects the library BusDevice and not core module busio, where there is a readfrom_into() method for i2c and readinto for spi.

tannewt, 2017-11-01:
@mrmcwethy Lets say 4.0 for removal.

What do you think about `readfrom_into`? It could be replaced as well. It'd just be a bit weird because it'd still take the address as a parameter.

mrmcwethy, 2017-11-03:
I am not changing busio interface at this time.  There already is a readinto method hidden in there as a primitive I2C operation.

mrmcwethy, 2017-11-05:
I am ready to release all the library changes.  Should I wait or start 12 pull requests?  Since these changes are incompatible with 2.x of circuitpython .  I assume we have a way to build/maintain library bundle 2.x and library bundle 3.0 separately.

tannewt, 2017-11-05:
Can we `try: except` it for now? That way it could be used on any version.

mrmcwethy, 2017-11-06:
here is a possible way to detect a missing method:

> 

    def _read_register(self, register, length):
        self._buffer[0] = register & 0xFF
        with self._i2c as i2c:
            i2c.write(self._buffer, start=0, end=1)
            try:
                i2c.readinto(self._buffer, start=0, end=length)
            except AttributeError:
                i2c.read_into(self._buffer, start=0, end=length)
            return self._buffer

>

deshipu, 2017-11-06:
I would rather put something like:
```
self.i2c_read_into = getattr(i2c, "readinto", getattr(i2c, "read_into"))
```
somewhere at the beginning (in the ``__init__``), so that we don't have to check every time.

mrmcwethy, 2017-11-06:
I like that idea @deshipu .  Thanks.

> 
class LIS3DH_I2C(LIS3DH):

    def __init__(self, i2c, address=0x18):
        import adafruit_bus_device.i2c_device as i2c_device
        self._i2c = i2c_device.I2CDevice(i2c, address)
        # handle name change of method readinto; backward compatible
        self._i2c_readinto = getattr(self._i2c, "readinto", getattr(self._i2c, "read_into", None))
        self._buffer = bytearray(6)
        super().__init__()

    def _read_register(self, register, length):
        self._buffer[0] = register & 0xFF
        with self._i2c as i2c:
            i2c.write(self._buffer, start=0, end=1)
            self._i2c_readinto(self._buffer, start=0, end=length)
            return self._buffer

>

deshipu, 2017-11-06:
I would drop that `None` from there — it should never get to that part, but when it does, and `AttributeError` is the right exception to throw.

mrmcwethy, 2017-11-06:
@deshipu  Python 3.5 evaluates both getattr() functions, so when we finally remove the redundant read_into method, the None ensures that the getattr(i2c,"read_into") does not throw an Attribute exception.  Maybe we need to add

> 

self.i2c_readinto = getattr(i2c,"readinto",getattr(ic2,"read_into",None))
# if both readinto and read_into are not present, then throw exception
 if self.i2c_readinto == None: raise AttributeError

> 

your thoughts?

deshipu, 2017-11-06:
Argh, you are right, I'm sorry, I didn't thought it through. In that case maybe:
```
try:
    self.i2c_readinto = i2c.read_into
except AttributeError:
    self.i2c_readinto = i2c.readinto
```
would be clearer.

tannewt, 2017-11-06:
The last option looks good! Also keep in mind that switching a driver to `adafruit_bus_device.I2CDevice` might be the easiest way going forwards.

mrmcwethy, 2017-11-07:
@tannewt I am not understanding your point about "switching a driver to".  Every one of the libraries I will be changing already are using adafruit_bus_device.I2CDevice.  What am i missing?

tannewt, 2017-11-07:
@mrmcwethy yeah, I was confused thinking we were changing CircuitPython's methods.

I think that we should just switch the libraries over to readinto. Lets leave read_into in the I2CDevice class for a little while for any other libraries to catch up.

I want people to grab libraries out of the bundle instead of separately because we can then verify bundled libraries always work with each other.

mrmcwethy, 2017-11-07:
I like that decision.  I will act on it.

mrmcwethy, 2017-11-08:
all bundle libraries now use readinto instead of read_into

tannewt, 2017-11-14:
Thanks @mrmcwethy!

closed 2017-11-14 by tannewt