adafruit/circuitpython #368: atmel-samd:  lto-type-mismatch and ERROR 1 on build

rrace001, 2017-10-25:
Output below.  I am using Ubuntu 16.04. I have compiled the UF2-SAMD21 without errors and flashed the bootloader.bin so my toolchain worked for that. (The bootdrive shows up on a double click but nothing else)  I just cloned circuitpython today and tried a few boards but all the same.  Typical output below.  I do not understand the error at all.

Output:
	/circuitpython/atmel-samd$ make V=1
GEN build-metro_m0_express/genhdr/qstr.i.last
cc1: error: -Werror=lto-type-mismatch: no option -Wlto-type-mismatch
	...prints this a couple hundred times...
cc1: error: -Werror=lto-type-mismatch: no option -Wlto-type-mismatch
../py/mkrules.mk:79: recipe for target 'build-metro_m0_express/genhdr/qstr.i.last' failed
make: *** [build-metro_m0_express/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build-metro_m0_express/genhdr/qstr.i.last'

jerryneedell, 2017-10-25:
After cloning repository do you do the following?
git submodule update --init --recursive

then when building it is good to d:
make BOARD=metro_m0_express clean
then
make BOARD=metro_m0_express

dhalbert, 2017-10-25:
Sounds like your toolchain is too old. You can get a newer one by doing this:
```
sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
sudo apt update
sudo apt install gcc-arm-embedded python build-essential
```

Take the old toolchain out of your path.

jerryneedell, 2017-10-25:
oops - that too!

hukuzatuna, 2017-10-25:
I had the same issue on Ubuntu under Windows 10. I can assert that updating the toolchain as dhalbert suggests, followed by the build process that jerryneedell documents, should solve the problem. If the OP (rrace001) can confirm the build process is working I think we can close this.

tannewt, 2017-10-25:
@rrace001 Please reopen if this is still an issue after updating your toolchain.

closed 2017-10-25 by tannewt

rrace001, 2017-10-25:
Thanks all.  Worked perfect!