adafruit/circuitpython #383: double file extensions

ladyada, 2017-10-28:
[image]

might be worth detecting this kinda thing at the 'end' of the list of files we run - windows tries to be 'smart' and not show extensions!

siddacious, 2017-11-02:
Do you mean we should run double extension files at a lower priority or just give a warning about them?

ladyada, 2017-11-02:
probably just run them at a lower priority - but a warning on REPL-open might be good

ladyada, 2017-11-02:
@siddacious if you're interested in taking this on, you can start by adding the list of files and we can do the warning later :)

siddacious, 2017-11-02:
Do we care about all the possible combinations or just doubling of the same extension?

dhalbert, 2017-11-02:
It would probably be good to make up a table of possibilities or iterate through the combinations, instead of having multiple calls to `maybe_run()`.

siddacious, 2017-11-02:
I've got the simple version but I'll take a whack at the warnings and combo iteration when I get home.

siddacious, 2017-11-06:
So I'm *so close* to being done with this except for the part where it's totally broken. When I take one of the generated double extension file names and run it (blinky in this case) with `pyexec_file(double_extension_filename, exec_result);` I see my led blink but I loose the CDC serial in /dev/tty* and CIRCUITPY doesn't show up. I assume it's running the blinky that I had previously uploaded and tested with master (5ff6) successfully since I don't have MSC to upload a new file with.

I'm going to try poking at it a bit but if I can't figure it out I'll commit to my fork so someone can point out what I'm doing wrong.

tannewt, 2017-11-06:
Is your branch based on 5ff6 as well? I'd be happy to take a look tomorrow.

siddacious, 2017-11-06:
I found the bug; I had not noticed that the two calls to `maybe_run` had different arguments and treated them as the same. I simplified how it works and addressed that oversight in doing so. Now I'm just trying to remember how to pointer.

commit b4fc464a (2017-11-08): support for doubled file extensions ala issue #383 (#415)

Fixes #383
closed 2017-11-08 by tannewt