adafruit/circuitpython #386: atmel-samd: File system erased on jittery power startup

tdicola, 2017-10-30:
So this goes back to the previous issues of losing the entire SPI flash filesystem on express boards in low power conditions.  With the latest 2.1 firmware I've noticed it unfortunately still occurs, just now I plugged in a JST battery to a cplay express with 2.1 and it erased the entire filesystem (was left with a blank root and boot_out.txt file).  Talking in chat one possible explanation is that jittery power on startup could cause a filesystem erase from brownout conditions.  This fits my repro.. when I was plugging in the battery this cplay express has a very tight JST connector and required a moment and extra firm wiggle to get seated properly.  During this time most likely the power was browing our or in an erratic state.  Perhaps should consider delaying a FS erase and checking voltage before it occurs?

tdicola, 2017-10-30:
This has been confirmed with another independent repro, from Kattni:

"@tdicola Repro'd.
I had to lightly touch the connector to the jack repeatedly before plugging it in to get it to repro. The status LED and D13 flashed each time I did that, and then when I plugged it in completely, it had erased."

Priority-wise I think this is worth looking at for the next 2.1 release.  It's super easy to misplug in a battery and could frustrate a lot of folks (losing all your code unexpectedly).

dhalbert, 2017-12-14:
I don't have a reliable fix for this yet. I tried adding a delay before reformatting the filesystem (so an immediate powerdown during the delay would avoid reformatting), and also delaying and re-checking when the filesystem seems bad. I can still bobble the JST connector and erase the filesystem.

dhalbert, 2017-12-19:
Going to put in the fix I have, because it's a lot better than nothing. PR #

dhalbert, 2018-01-09:
Close this for now, as the current fix seems to work pretty well. Reopen if it can be duplicated reliably in real-life situations.

closed 2018-01-09 by dhalbert

deshipu, 2018-01-10:
As a side note, personally I would love to have some way of triggering this "bug" on purpose — the solution with Arduino programs for erasing the flash doesn't work with custom boards, as I would need to make Arduino board definitions for them first, etc.

dhalbert, 2018-01-10:
@deshipu I have been thinking of adding something like `storage.format(...)`, with maybe some required `True` parameter to make clear it was dangerous.

deshipu, 2018-01-10:
Reminds me of the `--clobber-my-computer` option we have in some OpenStack tools. We could make it a required keyword argument, something like `storage.format(delete_all_files=True)` or such, but I wonder how much of a warning we actually need here?