adafruit/circuitpython #388: PDM mic buffer size causes hard reset and boot into safemode

AidenRay, 2017-10-31:
Running the following code from [readthedocs](https://circuitpython.readthedocs.io/en/latest/shared-bindings/audiobusio/PDMIn.html) causes a hard reset, and the CPX to run in safemode (not always).

```import audiobusio
import board

# Prep a buffer to record into
b = bytearray(200)
with audiobusio.PDMIn(board.MICROPHONE_CLOCK, board.MICROPHONE_DATA) as mic:
    mic.record(b, len(b)) ```

Increasing the buffer size to 8000 ('b = bytearray(8000)') fixes the issue.

AidenRay, 2017-10-31:
I tested a bunch of buffer sizes.

A length of 1504 is fine, whereas a length of 2963 hard resets. 
So simply increasing the buffer length doesn't seem to solve it.

dhalbert, 2017-12-14:
Fixed by #479.

closed 2017-12-14 by dhalbert