adafruit/circuitpython #395: ESP8266 - turn on framebuf in mpconfigport.h

dobra-dobra, 2017-11-01:
Framebuf is not included in ESP8266 port, which casuses incompatibility with SAMD port. Moreover, when framebuf is not included OLED screens can't be used to print strings.

deshipu, 2017-11-01:
You can install the Python implementation of framebuf, which should make all the drivers work (albeit slow): https://github.com/adafruit/micropython-adafruit-framebuf

jerryneedell, 2017-11-01:
The problem is that the python framebuf.py does not support text.

deshipu, 2017-11-01:
You can use other libraries to render text, such as my https://bitbucket.org/thesheep/font454 or Peter's https://github.com/peterhinch/micropython-font-to-py.git — there is an example for the latter at https://github.com/peterhinch/micropython-samples/blob/master/SSD1306/README.md

tdicola, 2017-11-01:
This is also a good option for rendering text without framebuf: https://learn.adafruit.com/micropython-displays-drawing-text  It's a library that you plug in a pixel drawing function so it can work with any kind of display (oled, neopixel matrix, etc.).

dhalbert, 2017-11-01:
I think there is room for framebuf in esp8266. If you want to turn it on now you could trying building it yourself (maybe you did that already). We would like to have it available everywhere if possible. We'll have to see what fits in 3.0 when we get closer to being done.

dobra-dobra, 2017-11-02:
Thank you for the suggestions. I'm aware that there are other options available, but in my opinion such small differences between ports are very annoying, especially when they are not clearly shown in docs. I use both SAMD and ESP8266-based boards and my point is that if there is no real need to keep such differences between ports they should be removed.

jerryneedell, 2017-11-05:
I'll be happy to submit a PR to enable frambuf  - Should I submit PRs both the 2.x and 3.0 branches?

dhalbert, 2017-11-05:
@jerryneedell Since it's already turned on in the atmel-samd express builds, sounds fine, assuming it fits. Do you have a way to test it for esp8266?

jerryneedell, 2017-11-05:
yes - I will test it with an ssd1306 on an esp8266 before submitting the PR. Likely tonight or tomorrow.

jerryneedell, 2017-11-05:
I have built it under both master and 2.x branches - no problem with she or import framebuf. Just need to hook up the oled display to verify operation.

jerryneedell, 2017-11-05:
submitted PR for 2.x  - Do you want one for master - I need to rebuild the .mpy files for master since the format is different. IS it useful to do that at this time. I was surprised that 3.0 even builds for esp8266, but it looks OK - I assume it is really the 2.x code. Let me know if you want me to test the .mpy builds and if you want a PR to master.

jerryneedell, 2017-11-06:
submitted  PR for master as well - I was able to generate the necessary .mpy's to test this in the master branch build using an ssd1306 OLED display.

jerryneedell, 2017-11-06:
Since the update is in 2.x and pending for master, I thing this can be closed. It won't let me close it.

tannewt, 2017-11-06:
I pulled the master version as well. Thanks @jerryneedell !

closed 2017-11-06 by tannewt