adafruit/circuitpython #401: 2017-11-02 3.0beta build REPL won't exit (M4)

hukuzatuna, 2017-11-02:
I did a git pull and submodule update this morning (2017-11-02) and flashed the firmware to the Metro M4 Express. Afterwards, I got a solid white NeoPixel.  When I connected to the REPL I got a prompt with no banner. The REPL works, though. When I hit control-D to exit the REPL it did a soft reboot and immediately re-entered the REPL. This happens repeatedly with control-D, regardless of whether or not I execute any Python in the REPL (not that I think it's likely that would change the behavior).

I did not yet try putting code.py on the filesystem to see if that changed the behavior (had to get to work).

This occurs on firmware built in the Unix environment on Windows 10 (second-to-latest Creator's update), then flashed to the Metro M4 Express with bossac from Windows.

hukuzatuna, 2017-11-03:
This morning's build (Adafruit CircuitPython 3.0.0-alpha.0-1143-ge7d6df7-dirty on 2017-11-03; Metro M4 Express with samd51j19) exhibits the same behavior, but I have some additional details.

- Flashing the new firmware results in a solid white Neopixel and a REPL that won't exit.
- There is no D:\CIRCUITPY (I'm on Windows) when the board is in that state.
- Pressing reset once gives me a pulsating green NeoPixel and a working REPL.
- Once I've done the above, the REPL works as expected wrt control-D and soft reboot.

I'll keep trying different combinations of things to get more information. If there's any other info you need, just let me know!

tannewt, 2017-11-07:
Can you test against latest master again? 4140b3b9443c47b8eda05de60b312b7df65ff5e0 fixed a usb issue and may have fixed this.

hukuzatuna, 2017-11-08:
I can't get to the REPL at all now. With Putty I can't open a serial connection to the COM port. I'm going to open a new issue with that problem. I will leave this one open since I don't know if the abovementioned USB issue has fixed this one.

mrmcwethy, 2017-11-23:
M4 status LED flashing with no main.py present #447 fixed a problem like the original issue.  When you don't have a code.py file, the board should display a slow blinking Green light.    Note the on Linux, I had to remove the ModemManger because that service was sending data to the board.  So on Linux, you may be getting sent to the REPL instead.  With my fix, the REPL should handle ctrl-D as expected. 
The issue on Windows sound like a different problem.  
Please retest with latest code.

dhalbert, 2018-01-09:
Appears to be fixed. Closing. Reopen if still a problem. Thanks!

closed 2018-01-09 by dhalbert