adafruit/circuitpython #406: release should include bin files for the uf2 based releases

zencuke, 2017-11-04:
Documentation says use bossac for recovery but that requires a .bin file. The .bin should be part of the release even if .uf2 is the primary. I assume the intent is to still support bossac.

tannewt, 2017-11-06:
Hi zencuke! Which documentation are you looking at? It should be updated.

I'd rather not include bins in the release list because its already long as is. However, they are always built and available here: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/

zencuke, 2017-11-07:
I was looking at the Circuit Playground Express tutorial. It is obviously not done. It is huge with lots of useful info but it looks like a lot of cut and paste was done. It needs some editing for things that need to be changed for inclusion in this specific tutorial. I try to give feedback when I notice things. It would work if the tutorial pointed to the bin files. It said "this is how to recover with a bin file" but no pointer to bin files. I went to the release area and no bin file.

zencuke, 2017-11-07:
This is off point for the issue but there was a bigger problem in the tutorial. I probably didn't need bossa. I tried to load circuitPython and the behavior didn't match the tutorial. Bright lights flashed, two popups appeared indicating that both CIRCUITPY and CPLAYBOOT had been improperly removed and CIRCUITPY was not on the desktop as promised.

It turns out that in spite of the weird behavior circuitPython was actually running. CIRCUITPY was mounted (according to 'df' and 'ls /Volumes/*') even though the icon did not appear on the desktop. However I didn't know that. I assumed the load failed and was trying to recover. That is the only reason I was looking at the bossa doc.

I eventually noticed that CIRCUITPY shows up on finder window sidebar but gave a strange error when selected. So it is somehow halfway mounted. It is good enough for bash access but the Finder gui sees it but doesn't like it.

zencuke, 2017-11-07:
I just tried the uf2 boot process again and much of the weirdness is gone. I only got a single unmount warning popup for CPLAYBOOT. None for CIRCUITPY this time. Even better CIRCUITPY was on the desktop. And the finder sidebar CIRCUITPY worked. I was able to drag the uf2 file to it.  If it had been doing this Saturday I wouldn't have been fiddling with bossa. I think I rebooted the Mac in there somewhere. OSX must have been in a bad state.

tannewt, 2017-11-07:
Cool cool! I've added a bin link to the guide here: https://learn.adafruit.com/adafruit-circuit-playground-express/circuitpython-detailed-install#flashing-with-bossac

What version of OSX are you on?

zencuke, 2017-11-07:
That works. You might mention that in the release notes as well going forward.

High Sierra 10.13. I have my Mac on auto update so I am usually pretty current. Except when I refuse to let it reboot of course. ;-)

tannewt, 2017-11-07:
Ok perfect! I've update the last release with a sentence and a link.

I'm also on the beta builds of High Sierra so I'll keep an eye out for it. For now, I'll close this issue. If you see the flakiness again please open another issue for it.

closed 2017-11-07 by tannewt