adafruit/circuitpython #407: reading value from output pin not always reflective

ladyada, 2017-11-05:
was testing a feather m0 basic on adafruit-circuitpython-feather_m0_basic-20171104-5ff6417 and bumped into this odd issue while testing pin toggling.

on some pins, reading the value does not return the right value?

e.g this code does not toggle:

```
led = digitalio.DigitalInOut(board.MOSI)
led.direction = digitalio.Direction.OUTPUT
while True:
    led.value = not led.value
    time.sleep(0.1)
```

but this works:

```
led = digitalio.DigitalInOut(board.MOSI)
led.direction = digitalio.Direction.OUTPUT
while True:
    led.value = True
    time.sleep(0.1)
    led.value = False
    time.sleep(0.1)
```

jerryneedell, 2017-11-06:
I've been staring at https://github.com/adafruit/circuitpython/blob/2.x/atmel-samd/common-hal/digitalio/DigitalInOut.c and I am wondering if switch_to_output() is not called then how is self->open_drain initialized? If it is not initialized, then will it behave differently for some pins?

If you just set the direction as in the example, I don't think switch_to_output() is ever called.

dhalbert, 2017-11-06:
`common_hal_digitalio_digitalinout_switch_to_output()` is called here: https://github.com/adafruit/circuitpython/blob/2.x/shared-bindings/digitalio/DigitalInOut.c#L202, in the layer above. When you set the `.direction` property it calls either `common_hal_digitalio_digitalinout_switch_to_input()` or `common_hal_digitalio_digitalinout_switch_to_output()`.

dhalbert, 2017-11-07:
This works fine on 2.x, but is a problem on 3.0.

Tested partially on Feather M0 basic and Metro M0:
MOSI, A1, A2, A5, value is False even after set to True.
MISO, A0:  works fine.

Then, while trying to use A3, I got a "Pin PB09" in use, even though I hadn't used A3 before. So this might be a pin mapping problem.

PR #416 (2017-11-07): DigitalInOut: Use pin value % 32 to accomodate PORTB and up.
Fixes #407 (get pin.value not correct for certain pins)
commit 9060df5a (2017-11-07): DigitalInOut: Use pin value % 32 to accomodate PORTB and up.
Fixes #407.
closed 2017-11-07 by tannewt