adafruit/circuitpython #417: 3..0: can only paste 128 chars into REPL

dhalbert, 2017-11-07:
Pasting from the clipboard or middle-click (Linux) into the REPL in 3.0 only allows 128 characters. In 2.x, much larger blocks can be pasted (I couldn't get it to break).

jerryneedell, 2017-11-07:
I'm glad you posted this - I had a cut/paste truncated recently on 3.0 and I was meaning to go back and explore it!

sommersoft, 2018-03-11:
So in a "lull", I found myself diving down this rabbit hole. I'm thinking that it's ATMEL/ASF USB related. I narrowed into `mphalport` and the stdin/stdout functions. ATMEL uses the ASF USB construct for interaction.

So, to test my budding theory, I did two things:
1. Tested pasting a long string in both regular REPL mode, as well as paste mode (Ctrl+E). Pasted string cut off at the same point (128 chars).
   - Full Test string: `x = "Testing 128 character pasting into the REPL to see if 3.0 still breaks when you try to do this. Blah blah, bleh bleh and stuff"`
   - Pasted result: `x = "Testing 128 character pasting into the REPL to see if 3.0 still breaks when you try to do this. Blah blah, bleh bleh and st"`
2. Booted up a Feather52, pasted the same string in regular and paste mode. Full string pasted in both modes.

I'll keep pulling on this string, but may come to you ASF gurus with some questions.

sommersoft, 2018-03-11:
Ok, so I don't think I'll have to get to the ASF level. But, here are more intermediate notes...

- `mphalport.c`: `mp_hal_stdin_rx_chr` & `mp_hal_stdout_tx_chr` use `usb_read()` and `usb_write()` respectively.
- `usb.c`: read and write functions rely heavily on constant `USB_RX_BUF_SIZE`
- `usb.h`: sets `USB_RX_BUF_SIZE = 128`

Might be _too_ coincidental, but I'll keep pulling. Really wish I had a JLink...

sommersoft, 2018-03-12:
Late night notes/sanity check, after beating `usb.c` into my brain and breaking it by using `printf`s.

Using the `USB_RX_BUF_SIZE = 128` thread I latched onto yesterday, I _think_ we're hitting the buffer limit on line 125 of `usb.c->read_complete()`:
```
if (count > (uint8_t) (USB_RX_BUF_SIZE - usb_rx_count)) {
        atomic_leave_critical(&flags);
        return true;
}
```
I got to this point with my crude `printf` debugging (no output; only breaking). When pasting under 128 chars, the `printf` wouldn't trip. Over 128...REPL became unresponsive (obviously).

Once I get a JLink in (on the way soon), I can verify this a little closer and begin working to a solution. I have a plan, that I may work a little before an actual debugger arrives.

**Plan**
The whole read process relies heavily on `USB_RX_BUF_SIZE`, `usb_rx_count`, and `count` to ensure we don't walk into a buffer overrun. I tried arbitrarily increasing `USB_RX_BUF_SIZE`...didn't go well. On my journey into the depths of ASF, I noticed that the structure that holds the `count` also has `size`. `count` is described as the "total transfered data count", which increments as each endpoint buffer is transferred (I think). `size` is described as the "total data size", which I'm **hoping** will allow us to use the `size` with `count` to determine `read_complete`. After some thought, my first approach will be to use `size` to calculate the size of an array of buffers, and then to keep track of looping and buffer placement.

And with that, I need to end this "congratulations you lost an hour just for waking up" day.

tannewt, 2018-03-12:
I think you are on the right track! I wrote the `ports/atmel-samd/usb.c` code so its not perfect. :-)

What does ASF4 do when true is returned?

The Feather nrf52 is a bad example because it uses a USB -> UART chip like the ESP8266. (We don't yet support the USB on the ones that have native USB.)

sommersoft, 2018-03-13:
> The Feather nrf52 is a bad example because it uses a USB -> UART chip like the ESP8266. 

When I initially glazed at the issue, I failed to recognize the "atmel-samd" label. I really just wanted to figure out if it was a core versus a port issue (with the knowledge that USB/UART implementations differ).

> What does ASF4 do when true is returned?

What I'm surmising is that it is releasing the lock on the received buffer [`usb_rx_buf`] and letting `usb_read()` return the data to `stdout`/REPL. 

Which of course is maxed out to the first 128 characters grabbed from the endpoint buffers. 

The only other time `read_complete == true` is when it checks the endpoint buffers for more data after processing the current buffer.

dhalbert, 2018-03-13:
2.x (ASF3) does not have this problem (I think), so that would an example of looking at how this is dealt with sucessfully.

sommersoft, 2018-03-13:
Just verified, 2.x does not have the problem. 

Quick glance at 2.x's equivalent higher level functions (all in `mphalport.c`; not broken out to a `usb.c`), and it seems to function in the same manner. From my previous googling, it seems there was a switch with how the ASF USB drivers operated between 3 & 4.

Having said that, EXCELLENT suggestion @dhalbert! Back into the abyss... :smile:

sommersoft, 2018-03-19:
Meant to put this up last night before the meeting today, but ran out of time.

- After looking at 2.x a little deeper, the only BIG difference I can see is that `usb_rx_notify` blocks out while it pulls the data from both endpoint buffers (128 bytes) and puts it in the local buffer. I may be wrong, but reconciling the difference between ASF3 and ASF4 conventions isn't easy for the untrained such as myself.

- 3.x `read_complete` is the equivalent [my perception] and only blocks out for each endpoint (64 bytes) at a time. `usb_read` which returns the data from the local buffer, begins reading the first 64 bytes, while `read_complete` begins to grab the second endpoint buffer. As I said earlier, I think this is where we're getting stuck. The three variables used to keep track of the transfer for the local buffer are `usb_rx_count`, `usb_rx_head`, and `usb_rx_tail`. `pending_read` is used to help ensure we don't step on communication with the endpoints, but as my debug logs are showing...this may be assisting in us getting stuck as well.

Here is an example of the logs I've gotten so far: [REPL Debug Gist](https://gist.github.com/sommersoft/fbfe8acae76589256f874fe63f697660)

I'm still working on getting clear debug data. Atmel Studio likes to crash & erase my bootloader... 🗡 
 Some of the finer points are also obscured by being optimized out during compile. I'm going to try and overcome that tonight, so we'll see. (Example: I can't debug anything from `read_complete`)

sommersoft, 2018-03-20:
New debug log: [REPL Debug Gist 2](https://gist.github.com/sommersoft/4b2623a69b6e606390d803492699c64b)

Here is an excerpt of the informational bits (better formatting here):

------------------------------------------------------------------------------------------------------------------
started this capture already at '>>>' (see first usb_rx_buf_tail == 70, with usb_rx_count == 64...difference of 6)

read_complete(_usb_d_cb_trans_done):: usb_rx_count: 0; pending_read: true     **#<-- first ep xfer**
start_read(read_complete):: usb_rx_count: 64; pending_read: false
usb_read(mp_hal_stdin_rx_chr):: usb_rx_count: 64; usb_rx_buf_head: 6; usb_rx_buf_tail: 70; pending_read: true

read_complete(_usb_d_cb_trans_done):: usb_rx_count: 64; pending_read: true   **#<-- 2nd ep xfer**
start_read(usb_read):: usb_rx_count: 127; pending_read: false

**read_complete(_usb_d_cb_trans_done):: usb_rx_count: 127; pending_read: true #<-- 3rd ep xfer**

------------------------------------------------------------------------------------------------------------------
This is the third transfer which would contain bytes 128 thru 192. `read_complete` has a conditional in the beginning to check for available buffer space: `if current endpoint data size [count] > (total buffer size [128] - usb_rx_count) { return true }`
 
I still need to narrow down where `read_complete`'s return goes. I have a suspicion that its returning to `_usb_d_cb_trans_done`, but I'll need to dig back into the ASF level. Following that logic, upon a return of true `_usb_d_cb_trans_done` may be responding back to the host with an ACK, which clears the ep buffer (false positive)

My only other theory is that once `usb_read` gets done reading, it never goes back to read the ep buffers, and the buffer clears after the timeout (500ms if my USB Nutshell reading is correctly comprehended). Still need to figure out why `pending_read` never clears back to false; can't get the debugger to list if any error codes are coming back in `start_read[result]` or `read_complete[rc]`. Atmel Studio and it's GDB terminal will only return <optimized out> no matter how I build the ELF.

tannewt, 2018-03-20:
@sommersoft I think you are on the right trail! We're trying to rate limit incoming data by erroring on the callback with the new data but `_usb_d_cb_trans_done` ignores the result.

Perhaps changing [this line](https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/usb.c#L155) to the below would fix it:

```c
if (usb_rx_count < USB_RX_BUF_SIZE - 64) {
```

That way we'd only queue up a read when we have room for another packet.

sommersoft, 2018-03-20:
@tannewt that might actually work. Initial thought is that it may force `usb_read` to clear the local buffer before triggering any further ep reads. Will likely have to change the `if (!pending_read && usb_rx_count == (USB_RX_BUFF - 1))`, unless we just wait on `usb_cdc_background` to call `start_read`. (On phone, by memory; otherwise I'd link to lines)

sommersoft, 2018-03-22:
Unfortunately, changing that line didn't work. 😞 Also disabled all of the 'in function' calls to `start_read` to see if letting `usb_bytes_available` would just pick up the additional transfers...nope. I finally put in some variables to extract the function args (`read_complete(return code)`, etc). Nothing is showing any issues just yet. So it may be deeper in the endpoint setup after all. Just gotta keep pushing...

tannewt, 2018-03-22:
What did it do? Do you have a copy of the changes? I expected it to work. :-)

sommersoft, 2018-03-22:
@tannewt, by changing the line to your suggested...it didn't do anything different. still 3 calls to `read_complete` but only printed out 128 characters.

When I was disabling 'in function' calls to `start_read`, I was trying to see if any of them were stepping on each other, in addition to seeing if `usb_bytes_available` would keep the wheel rolling. Oddly, it reduced the instances of `read_complete` from 3 to 2, but still produced the same 128 characters.

Baffling is the only word I can muster, although I'm still on about 70% confident I understand both how it is working and how it is supposed to work.

To make matters worse, I did see a couple pasted strings that went beyond 128 characters. I thought that they were happening because I was continuing to hold the paste keys longer...but couldn't reproduce with any consistent results.

sommersoft, 2018-03-22:
I'm moving on to the `usb_write/mp_hal_stdout` side. I'm getting more and more convinced we're chasing a red herring on the read side. 

Just spent the last 30 minutes stepping in/over `read_complete`s while pasting and watching the callstack. This seems to be a little more accurate in Atmel Studio than printing out vars and automatically continuing execution.

sommersoft, 2018-03-31:
Fixed by #698.

closed 2018-03-31 by sommersoft