adafruit/circuitpython #423: corrupted file system when exceeding capacity on 3.0 - M4

jerryneedell, 2017-11-08:
I ran into an issue last night when I mistakenly tried to load the whole bundle lib/ onto my Metro-M4 with the current master 3.0 build.  There was not enough room, but I did not receive any "out of space" errors. instead, the FS became corrupted and would only mount as read-only. Several files were clearly corrupte (truncated). I eventually erased the whole flash and restored it with only the necessary libraries and it was all working again.

I have seen out of space errors in the past with 2.x on the non-express boards (trinket m0) without corrupting the FS. I have not tested this recently. 

I just wanted to open this issue as a warning and will try to do some more testing on both 2.x and 3.0 soon.

jerryneedell, 2018-03-04:
I suggest we close this as OBE (Overcome By Events) - It was initiated before the Flash FS was enabled and never reproduced.   I just tried overfilling the Flash FS several times and it was handled gracefully on my Metro M4 Express Revb with current master build.

tannewt, 2018-03-05:
Perfect! Thanks for filing it in the first place and circling back.