adafruit/circuitpython #439: No support for Feather/FeatherWings w/ RFM95 LoRa and RFM69

BitKnitting, 2017-11-16:
Feature request: 
Awesome: send / receive RFM69 and RFM95 LoRa.  
OK: send/receive RFM95 LoRa 
Minimum: Support for ISRs so a person can build a library.
CircuitPython is awesome.  Sadly, the Huzzah is the only Feather who can fly.  This is ironic, since the simplicity of the RFM96 and RFM69 make it ideal for the beginner who wants to add wireless to their project.  For example, the good 'ol "You've Got Mail" mailbox open detector that sends an RFM 95 LoRa into the excited circuitpythonista's house.

siddacious, 2017-11-16:
RMF97/69 support would be cool but ISRs in general would be great to have.

If they're not currently supported, this page is slightly misleading: [ISR page on readthedocs](http://circuitpython.readthedocs.io/en/2.x/docs/reference/isr_rules.html)

I'm guessing that it's automatically rendered from micropython code that isn't used but that certainly isn't clear if thats the case.

tannewt, 2017-11-16:
Why do we need ISRs as opposed to polling? Concurrency such as ISRs is complex.

@siddacious I agree we should remove that ISR doc page.

siddacious, 2017-11-17:
My thought was that ISRs would free up some resources from polling, though honestly I don't know what resources those would be. Clock cycles? Maybe that's not an issue.

I suppose my thought was that CP should be able to do the normal MCU things that Arduino and others can and that people willing to pay the complexity cost would do so. Then again I don't know how much people would want use it or what the development opportunity cost would be.

tannewt, 2017-11-17:
Yeah, I don't want to assume our primary audience is people with existing Arduino and microcontroller experience. Instead, I want to approach it from a blank slate perspective and there isn't an easy to understand way of doing concurrency.

One thing mark shannon and I were kicking around was a way of doing an event loop where you can polling for a number of separate events simultaneously.

sKr0d, 2017-12-04:
I would like to +1 support for RFM69.  I have a lot of projects using RFM69 on Arduino code, and would love to see some wings on my Feather M0 RFM69HCW!!   I'm not a driver coder, but I understand Python and could contribute to testing, since I have a bunch of RFM69 boards and some projects.

siddacious, 2017-12-12:
@tannewt I was pointed to the ISR doc page and can remove it; should I create a new issue for cleaning up it and similar docs?

tannewt, 2017-12-12:
New issue would be great!

samaparicio, 2018-01-02:
+1 for RFM69 / RFM95 support - this is a very useful building block and today, if you want to use Lora then you are not going to use python.

tannewt, 2018-01-16:
Basic RFM69 support is here: https://github.com/adafruit/Adafruit_CircuitPython_RFM69

Since we don't have ISR support it only supports one packet at a time.

ladyada, 2018-01-24:
We've also got a binary release for the Feather M0 RFM69's w/CircuitPy 2.2.1 so you can use them as-is for radio projects. We may attack RFM9X next. There's no chance we will fit LoRaWAN in the M0 Py build but we may be able to send/receive LoRa packets (stay tuned! :)

BitKnitting, 2018-02-27:
I am gleefully using the RFM69.  Thank you.  I have a greater need for the RFM95 LoRa - so excited to read the above comment.  With that said, again - THANK YOU for the RFM69 driver.  There is a lot to learn from going through Tony's code.

ladyada, 2018-02-27:
yep! we're working on it, if you can help test that would b awesome :) note we wont have lorawan support - there's just no way to fit it (yet)

tdicola, 2018-03-06:
Yep check out the new guide page here for info on the new library: https://learn.adafruit.com/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts/circuitpython-for-rfm9x-lora  It's very similar to the RFM69 library but made to work with the RFM9x radios in LoRa mode and with RadioHead compatible headers/config.  Basic send/receive between two modules running the CircuitPython code or the RadioHead Arduino library RFM95 example (basic 'raw' client / server) work great.

closed 2018-03-06 by tdicola

tannewt, 2018-09-07:
@WRadigan That is outdated. The latest docs are here: https://circuitpython.readthedocs.io/en/3.x/