adafruit/circuitpython #447: M4 status LED flashing with no main.py present

kattni, 2017-11-17:
The status NeoPixel flashes an error code when no `main.py` is present on the board. It flashes blue for ~0.5s, yellow for ~0.5s, and then it quickly flashes some more yellow potentially followed by cyan. It is similar to the type of sequence that usually indicates the line on which an error is located in `main.py`. 

This happens regardless of how `main.py` is removed. I performed an erase and clean flash of the current 3.0 build. I also loaded, ran and deleted a `main.py` from the drive without a clean flash. 

It does not flash the same sequence each time - although repeatedly following both sets of steps seems to reveal one of two flash sequences (blue, yellow, and flashing yellow 6x // blue, yellow, flashing yellow 5x then cyan ~9x). I thought I remembered there being more than just those two, but current repeated testing only results in those sequences.

Steps followed in renaming:
1. Load and run a `main.py`.
2. Rename `main.py` to something else.
3. Wait for the error to flash.
4. Rename it back to `main.py`.
5. Repeat.

When there was no `main.py`, this typically resulted in the blue-yellow-yellowflash-5x-cyanflash-9x sequence.

Steps followed in clearing board completely:
1. Erase the board completely (using the M4 eraser).
2. Reflash CircuitPython onto the board from the current build.

The second set of steps results immediately in the status LED - typically the blue-yellow-yellowflash-6x sequence.

jerryneedell, 2017-11-20:
Interesting - o few observations:
after a power cycle
I am seeing a flashing green pattern when no main.py is present.
Solid green when main.py is present.

however if I have a main.py then remove main.py it goes into  a very colorful  fast flashing sequence when it reboots or after pressing RESET.

It goes back to flashing green if do do a power cycle.

On at lease one occasion it go int a state when it came up white (as in REPL) but screen was not connected. It went back to flashing green when I disconnected the USB cable (It is powered by 9v)

tannewt, 2017-11-20:
If anyone wants to take a look at this I'd suggest starting here:

https://github.com/adafruit/circuitpython/blob/master/main.c#L161
https://github.com/adafruit/circuitpython/blob/master/supervisor/shared/rgb_led_status.c#L202

mrmcwethy, 2017-11-21:
Now that the clock freq is fixed, i compared the behavior of Metro M0 2.x against Metro M4 3.x.  They seem to have the same behavior with respect to the neopixel.  Now the question is what should the neopixel behavior be? 

- solid green when running
- solid white when in REPL
- Green-Yellow-Blue-Blue-etc when Press any key to enter the REPL. Use CTRL-D to reload.

tannewt, 2017-11-21:
@mrmcwethy when it gets to the "Press any key.." line it should fade green in and out if no error occurred while running main.

Yup, I'm aware the status neopixels are turned off and just haven't gotten to it. Mind taking a look? It should be as simple as uncommenting things in mpconfigport.h IIRC.

mrmcwethy, 2017-11-21:
What I am seeing is with *no* main.py file then after a hard reset, the neopixel shows solid White (which means board is in the REPL) and indeed the board is placed in the REPL.  This is what the Metro M0 2.x and the Metro M4 3.x boards do.

The only odd thing is when you rename the main.py to mainX.py (you have to wait quite awhile)  This is where the multiple flashing occurs.  If you open the serial terminal there was a KeyboardInterrupt at line 61, then a couple of auto-reboot messages, and finally a "Press any key to enter the REPL. Use CTRL-D to reload." message.  The goes long Blue, long Yellow, then 6 short Yellow and 1 short Blue.  Even though there isn't a main.py, it remembers the KeyboardInterrupt exception and blinks that line number.

In the case of no main.py, we should either go to the REPL (like the hard Reset) or we should show pulsing Green as described below.

I found the following from Metro M0 Help:
```

When CircuitPython is running, what do the colors and blinking mean on the RGB status LED?

The Feather M0 Express, Metro M0 Express, Gemma M0, and Trinket M0 all have a single NeoPixel or DotStar RGB LED on the board that indicates the status of CircuitPython. Here's what the colors and blinking mean:

    steady GREEN: main.py (or main.txt, code.py, or code.txt) is running
    pulsing GREEN: main.py has finished or does not exist
    YELLOW: Circuit Python is in safe mode: it crashed and restarted
    WHITE: REPL is running
    BLUE: Circuit Python is starting up

Colors with multiple flashes following indicate a Python exception and then indicate the line number of the error. The color of the first flash indicates the type of error:

    GREEN: IndentationError
    CYAN: SyntaxError
    WHITE: NameError
    ORANGE: OSError
    PURPLE: ValueError
    YELLOW: other error

These are followed by flashes indicating the line number, including place value. WHITE flashes are thousands' place, BLUE are hundreds' place, YELLOW are tens' place, and CYAN are one's place. So for example, an error on line 32 would flash YELLOW three times and then CYAN two times. Zeroes are indicated by an extra-long dark gap.
```

mrmcwethy, 2017-11-21:
I updated the Metro M0 Express and Feather M0 Express board to enable Neopixel status light.

tannewt, 2017-11-21:
@mrmcwethy we should go pulsing green. If we go straight to REPL then we'll turn off the autoreload.

mrmcwethy, 2017-11-22:
@tannewt Just to clarify: change circuitpython do that it does not do an auto-reload when there is no main.py?  We should be seeing a blinking Green neopixel.

tannewt, 2017-11-22:
No, auto-reload should be enabled and the led fading green in and out.

mrmcwethy, 2017-11-22:
I am able to improve the soft reboot case (renaming a file) by "zeroing" out a stack variable right after a declaration.  I don't know if there is a compiler option to zero stack allocations, but after writing code to zero, the Blinking Green is now working.

main.c start_mp()
```
    pyexec_result_t result;
    result.return_code = 0;
    result.exception_type = NULL;
    result.exception_line = 0;
```
Have you had this problem in other circumstances?

mrmcwethy, 2017-11-22:
With my changes above, I am seeing a difference in behavior from Ubuntu 16.04 and Windows 10.  On Windows 10 if no main.py exists, you get the blinking Green neopixel.  On Unbuntu, you get the solid White (which means in REPL).  There is some code that checks the serial port for any available.  On Ubuntu, this code I believe is return true, while on Windows 10 I believe its false.  I think we have a work status light on Windows 10.  

I need to get a debugging system going to go further.  Is there a tutorial on setting up debugging for circuitpython?  I have a j-link cable.  I not sure what to do with it.

dhalbert, 2017-11-22:
https://learn.adafruit.com/debugging-the-samd21-with-gdb (by @tannewt )

microbuilder, 2017-11-22:
There are some further tips on debugging via GDB here, even though this particular page is aimed at a different platform (Apache Mynewt running on the nRF52): https://learn.adafruit.com/adafruit-nrf52-pro-feather/gdb-debugging ... but the commands are identical across devices. Just as a FYI!

tannewt, 2017-11-23:
@mrmcwethy I believe the white pixel on Ubuntu is because of ModemManager trying to talk with it. I disabled ModemManager on my Rosie box.

mrmcwethy, 2017-11-23:
@tannewt  Yes, I remove ModemManager and now the board is consistant with Window 10.  If there is no main.py, the board show a slow blinking Green neopixel.  I am going to release my changes and hope to close this issue.

mrmcwethy, 2017-11-23:
Fixed in circuitpython/master

closed 2017-11-23 by mrmcwethy