adafruit/circuitpython #448: Add audiobusio to the Support Matrix in the doc

dhalbert, 2017-11-18:
http://circuitpython.readthedocs.io/en/2.x/shared-bindings/index.html

sommersoft, 2018-01-01:
@dhalbert in reference to the comments on PR #503, is this along the lines of what you're looking at moving too? Kind of a middle ground of the current table, and listing only exceptions of unsupported ports.

This is going off the current list, and carbon copying SAMD51 to current SAMD21 support (which isn't an accurate future state given 51's higher mem capacity). Its right at 50/50 for supported/unsupported, so a toss-up between current style and exception-only style IMO.

I'm going to drop this into my fork on the 3.x branch...

```
Support Matrix
---------------
**All Supported** means the following ports are supported: SAMD21, SAMD21 Express
(SAMD21X), SAMD51, SAMD51 Express (SAMD51X), and ESP8266. [nRF5x to be added later]

=================  ===============
Module             Supported Ports  				
=================  ===============
`analogio`         **All Supported**
`audiobusio`       **All SAMD/SAMDX**
`audioio`          **SAMD21X, SAMD51X**
`bitbangio`        **SAMD21X, SAMD51X, ESP8266**
`board`            **All Supported**
`busio`            **All Supported**
`digitalio`        **All Supported**
`gamepad`          **SAMD21X, SAMD51X**
`math`             **All Supported**
`microcontroller`  **All Supported**
`multiterminal`    **ESP8266**
`neopixel_write`   **All Supported**
`nvm`              **SAMD21X, SAMD51X**
`os`               **All Supported**
`pulseio`          **All SAMD/SAMDX**
`random`           **All Supported**
`storage`          **All Supported**
`struct`           **All Supported**
`supervisor`       **All SAMD/SAMDX**
`time`             **All Supported**
`touchio`          **All SAMD/SAMDX**
`uheap`            **Debug (All)**
`usb_hid`          **All SAMD/SAMDX**
=================  ===============
```

dhalbert, 2018-01-01:
Yes, thanks, something like that. I'd say "Express" explicitly, instead of introducing a new abbreviation ("X").

tannewt, 2018-01-16:
This is done. Thanks @sommersoft !

closed 2018-01-16 by tannewt