adafruit/circuitpython #457: permission error for lib/nrfutil

jerryneedell, 2017-11-23:
after gitt pull for latest master
git submodule update --init --recursive fails for lib/nrfutil due to a permissions error:

```
$ git submodule update --init --recursive 
Submodule 'lib/nrfutil' (git@github.com:adafruit/nRF52_nrfutil.git) registered for path 'lib/nrfutil'
Cloning into 'lib/nrfutil'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:adafruit/nRF52_nrfutil.git' into submodule path 'lib/nrfutil' failed```

a workaround is to change line 45
```url = git@github.com:adafruit/nRF52_nrfutil.git
to
url = https://github.com/adafruit/nRF52_nrfutil.git```

should this be the permanent fix or should the permissions be changed?

dhalbert, 2017-11-23:
This should be permanent. All submodules should always be https://github.com/...

microbuilder, 2017-11-23:
I'll switch to https instead of git@ for the PR. Didn't realize this would cause a permissions error, thanks for the heads up.

closed 2017-11-23 by microbuilder

tannewt, 2017-11-23:
FYI I changed master's settings to only allow @dhalbert and I to push directly to it. Please ping us if cleanup is needed in the future.