adafruit/circuitpython #458: nrf/feather52 build fails

jerryneedell, 2017-11-23:
attempting to build:
 make BOARD=feather52 V=1 SD=s132
 results in:
```
FREEZE freeze
Creating build-feather52-s132/frozen_mpy.c
CC build-feather52-s132/frozen_mpy.c
CC ../../main.c
../../main.c: In function 'nlr_jump_fail':
../../main.c:343:5: error: implicit declaration of function 'HardFault_Handler' [-Werror=implicit-function-declaration]
     HardFault_Handler();
     ^~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
../../py/mkrules.mk:53: recipe for target 'build-feather52-s132/main.o' failed
make: *** [build-feather52-s132/main.o] Error 1
```

jerryneedell, 2017-11-23:
I tried just disabling the warning above but things deteriorate...
results in:
```FREEZE freeze
Creating build-feather52-s132/frozen_mpy.c
CC build-feather52-s132/frozen_mpy.c
CC ../../main.c
CC supervisor/port.c
In file included from supervisor/port.c:26:0:
../../supervisor/port.h:44:8: error: unknown type name 'uint32_t'
 extern uint32_t _ezero;
        ^~~~~~~~
../../supervisor/port.h:47:8: error: unknown type name 'uint32_t'
 extern uint32_t _estack;
        ^~~~~~~~
../../supervisor/port.h:50:8: error: unknown type name 'uint32_t'
 extern uint32_t _ebss;
        ^~~~~~~~
In file included from supervisor/port.c:27:0:
./boards/board.h:34:17: error: unknown type name 'uint32_t'
 extern volatile uint32_t ticks_ms;
                 ^~~~~~~~
../../py/mkrules.mk:53: recipe for target 'build-feather52-s132/supervisor/port.o' failed
make: *** [build-feather52-s132/supervisor/port.o] Error 1
```

dhalbert, 2017-11-23:
Some file should be doing `#include <stdint.h>` but isn't.

jerryneedell, 2017-11-23:
if I clone the repository from https://github.com/microbuilder/circuitpython then the build succeeds.
Perhaps somehting got missed in the commit to adafruit/circuitpython

dhalbert, 2017-11-23:
@microbuilder pushed to adafruit/circuitpython instead of his own repo by mistake. He'll revert this and then resubmit as a PR including the fix.

microbuilder, 2017-11-23:
The last commit was just reverted, sorry for the trouble.

closed 2017-11-23 by microbuilder

jerryneedell, 2017-11-23:
Thanks - No problem - I have cloned your repo and it is working - looking forward to having it here as well.