adafruit/circuitpython #463: Circuit Playground Express reports NO NAME after flash

jadudm, 2017-11-27:
Hi team,

I have a Playground Express purchased... in August, I think (in case that makes a difference). This is on a Mac running 10.12.6. The USB cable works with Micro:Bits, and appears to work with the CPE.

My process:

1. Double-click to get into bootloader mode. The CPE appears as CPLAYBOOT, as expected.
2. I have pulled the current UF2 file from the tutorial, and drag it onto the CPE. This appears to transfer correctly.
3. CPLAYBOOT disappears.
4. NO NAME appears.

This drive is, apparently, a zero-byte FAT-formatted drive. I cannot save Python files to it. Resetting does nothing. I can reboot into bootloader mode, and (at that point), successfully upload code from Makecode. Or, if you prefer, code generated from Makecode correctly transfers over, but the CircuitPython build does not?

I have not had any success Googling this issue, nor have I found anything in the Github issues for CP/CPE, so I reported it here. If there is a better forum for exploring this bug (which may be a user bug, as opposed to a software bug), please let me know. My next step will be to attempt to use the bossac tool to program the CPE, and see if that gives me different results.

deshipu, 2017-11-27:
Hi, have you tried this fix? https://circuitpython.readthedocs.io/en/latest/docs/troubleshooting.html#file-system-issues

jadudm, 2017-11-27:
I had not been able to find that documentation. It fixed the CPE; thank you.

I will mark this closed, and find somewhere in the documentation where I think this should be mentioned. Then, I'll fork, branch, edit, merge trunk, and make a pull request against that documentation edit from my branch. I'm assuming it will be in one or two places in the CPE tutorials that this would be good to mention. (And, assuming those tutorials/documentation are in Git somewhere... if not, I'll just report it as best I can to somewhere/someone at Adafruit. Either way, I can do that digging.)

(And, it is possible it was mentioned, and I missed it.)

closed 2017-11-27 by jadudm

deshipu, 2017-11-27:
Don't worry, I had problems finding it myself. I think part of the problem is that it's only in the latest documentation, and not in the "stable" one.

tannewt, 2017-11-27:
@jadudm If you find a spot it should go in an Adafruit learn guide feel free to ping me. They are not on github.

JnBrymn, 2019-08-01:
Hi @deshipu I appear to be having the same problem as @jadudm described in the initial comment.  Reviewing the instructions you referenced (https://circuitpython.readthedocs.io/en/latest/docs/troubleshooting.html#file-system-issues) they ask me to "Download the appropriate flash .erase uf2 from the Adafruit_SPIFlash repo." and linked here https://github.com/adafruit/Adafruit_SPIFlash/tree/master/examples/flash_erase_express. Unfortunately there are no uf2's there.

Are there updated instructions?

JnBrymn, 2019-08-01:
Ah found it! https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#old-way-for-the-circuit-playground-express-feather-m0-express-and-metro-m0-express-20-25