adafruit/circuitpython #470: Clicks and buzzing with audioio.AudioOut

deshipu, 2017-12-04:
I just tried to make the first game with sound for µGame, and I'm struggling with it a little. I worked around a number of problems, but I'm still left with two issues when playing `.wav` files with `AudioOut`:
- There are loud clicks at the beginning and at the end of the samples. As I understand, those come from the sound file not starting/ending with a neutral value (middle of the range), and I should be able to solve it by processing the sound files to fix it. However, I wonder if it would be possible for the `AudioOut` code to deal with it internally.
- Sometimes a short file will never finish playing, but instead stop at the end with a loud buzzing sound, and stay there forever. Calling `.stop()` on it makes it stop. It happens randomly, and doesn't depend on the file itself.
I'm still investigating what is happening exactly, and I will keep updating this issue.
Here's a video showing the sound problems (the clicks are from the speaker, not from pressing the buttons):

https://youtu.be/fCqvTueoG7I

deshipu, 2017-12-04:
I got rid of the clicks, but I still can't figure out the problem with the buzzing. It seems like the last few samples are stuck playing at the end.

deshipu, 2017-12-04:
I fugured out the buzzing. Seems like the files that sfxr generates are truncated.

tannewt, 2017-12-05:
@deshipu can this be closed? It seems like you fixed it by changing the audio files.

deshipu, 2017-12-05:
Yes, I fixed my issues, however, I still plan to look at the `audioio` code to see if it could be made more robust. I'm not sure if some sort of smoothing between the last sample and next sample is feasible, but I think I can definitely handle the truncated files better.

ladyada, 2021-03-24:
closing for lack of activity

closed 2021-03-24 by ladyada