adafruit/circuitpython #475: Lint all the libraries!

tannewt, 2017-12-07:
I'm going through and updating the libraries to our new build process and to turn on lint. Here is the process:

1. Copy `.pylintrc` from the cookiecutter repo into the new repo.
2. Add this to `.gitignore`
```
.env
build*
bundles
```
3. Copy .travis.yml from cookiecutter into the repo overwriting the old version.
4. Update lines 1 and 3 of the `script` section of `.travis.yml` to reflect the driver name.
5. Add this to all modules in the driver with the correct repo url below the module comment:
```
__version__ = "0.0.0-auto.0"
__repo__ = "<repo url"
```
6. Run pylint on all modules and fix all errors.
7. Run `pylint --disable=missing-docstring,invalid-name examples/*.py` to pylint all examples and fix all errors.

If you are doing the PR then please check the box below before you start! That way no one else will do it at the same time.

## Remaining
- [ ] [Adafruit_CircuitPython_SSD1306](https://github.com/Adafruit_CircuitPython_SSD1306)
- [ ] [Adafruit_CircuitPython_RTTTL](https://github.com/Adafruit_CircuitPython_RTTTL)
- [ ] [Adafruit_CircuitPython_Waveform](https://github.com/Adafruit_CircuitPython_Waveform)
- [ ] [Adafruit_CircuitPython_HID](https://github.com/Adafruit_CircuitPython_HID)
- [ ] [Adafruit_CircuitPython_MAX7219](https://github.com/Adafruit_CircuitPython_MAX7219)
- [ ] [Adafruit_CircuitPython_SD](https://github.com/Adafruit_CircuitPython_SD)
- [ ] [Adafruit_CircuitPython_IS31FL3731](https://github.com/Adafruit_CircuitPython_IS31FL3731)
- [ ] [Adafruit_CircuitPython_OneWire](https://github.com/Adafruit_CircuitPython_OneWire)
- [ ] [Adafruit_CircuitPython_SimpleIO](https://github.com/Adafruit_CircuitPython_SimpleIO)
- [ ] [Adafruit_CircuitPython_PCA9685](https://github.com/Adafruit_CircuitPython_PCA9685)
- [ ] [Adafruit_CircuitPython_Thermistor](https://github.com/Adafruit_CircuitPython_Thermistor)
- [ ] [Adafruit_CircuitPython_seesaw](https://github.com/Adafruit_CircuitPython_seesaw)
- [ ] [Adafruit_CircuitPython_IRRemote](https://github.com/Adafruit_CircuitPython_IRRemote)
- [ ] [Adafruit_CircuitPython_RGB_Display](https://github.com/Adafruit_CircuitPython_RGB_Display)
- [ ] [Adafruit_CircuitPython_VC0706](https://github.com/Adafruit_CircuitPython_VC0706)
- [ ] [Adafruit_CircuitPython_BMP280](https://github.com/Adafruit_CircuitPython_BMP280)
- [ ] [Adafruit_CircuitPython_MAX31855](https://github.com/Adafruit_CircuitPython_MAX31855)
- [ ] [Adafruit_CircuitPython_SI7021](https://github.com/Adafruit_CircuitPython_SI7021)
- [ ] [Adafruit_CircuitPython_BME680](https://github.com/Adafruit_CircuitPython_BME680)
- [ ] [Adafruit_CircuitPython_GPS](https://github.com/Adafruit_CircuitPython_GPS)
- [ ] [Adafruit_CircuitPython_BNO055](https://github.com/Adafruit_CircuitPython_BNO055)
- [ ] [Adafruit_CircuitPython_INA219](https://github.com/Adafruit_CircuitPython_INA219)
- [ ] [Adafruit_CircuitPython_TSL2561](https://github.com/Adafruit_CircuitPython_TSL2561)
- [ ] [Adafruit_CircuitPython_BusDevice](https://github.com/Adafruit_CircuitPython_BusDevice)
- [ ] [Adafruit_CircuitPython_CCS811](https://github.com/Adafruit_CircuitPython_CCS811)
- [ ] [Adafruit_CircuitPython_DS3231](https://github.com/Adafruit_CircuitPython_DS3231)
- [ ] [Adafruit_CircuitPython_MPR121](https://github.com/Adafruit_CircuitPython_MPR121)
- [ ] [Adafruit_CircuitPython_PCF8523](https://github.com/Adafruit_CircuitPython_PCF8523)
- [ ] [Adafruit_CircuitPython_Register](https://github.com/Adafruit_CircuitPython_Register)
- [ ] [Adafruit_CircuitPython_ADS1x15](https://github.com/Adafruit_CircuitPython_ADS1x15)
- [ ] [Adafruit_CircuitPython_AMG88xx](https://github.com/Adafruit_CircuitPython_AMG88xx)
- [ ] [Adafruit_CircuitPython_CharLCD](https://github.com/Adafruit_CircuitPython_CharLCD)
- [ ] [Adafruit_CircuitPython_LIS3DH](https://github.com/Adafruit_CircuitPython_LIS3DH)
- [ ] [Adafruit_CircuitPython_NeoPixel](https://github.com/Adafruit_CircuitPython_NeoPixel)
- [ ] [Adafruit_CircuitPython_VL53L0X](https://github.com/Adafruit_CircuitPython_VL53L0X)
- [ ] [Adafruit_CircuitPython_TCS34725](https://github.com/Adafruit_CircuitPython_TCS34725)
- [ ] [Adafruit_CircuitPython_BME280](https://github.com/Adafruit_CircuitPython_BME280)
- [ ] [Adafruit_CircuitPython_DS18x20](https://github.com/Adafruit_CircuitPython_DS18x20)

## Pending PRs
These PRs may need testing before they can be merged.
- [Adafruit_CircuitPython_LSM9DS1](https://github.com/Adafruit_CircuitPython_LSM9DS1)
- [Adafruit_CircuitPython_LSM9DS0](https://github.com/Adafruit_CircuitPython_LSM9DS0)
- [Adafruit_CircuitPython_FXOS8700](https://github.com/Adafruit_CircuitPython_FXOS8700)
- [Adafruit_CircuitPython_FXAS21002C](https://github.com/Adafruit_CircuitPython_FXAS21002C)
- [Adafruit_CircuitPython_MAX31865](https://github.com/Adafruit_CircuitPython_MAX31865)
- [Adafruit_CircuitPython_HT16K33](https://github.com/Adafruit_CircuitPython_HT16K33)
- [Adafruit_CircuitPython_MCP9808](https://github.com/Adafruit_CircuitPython_MCP9808)

tannewt, 2017-12-07:
Background on the lint configs is available here: https://github.com/adafruit/cookiecutter-adafruit-circuitpython/pull/6

mrmcwethy, 2017-12-08:
I will start with MAX7219

mrmcwethy, 2017-12-09:
I added AdaFruit_CircuitPython_DHT to list and then converted it.

mrmcwethy, 2017-12-09:
I may have lost changes by ladyada and katni.  changed your comment for this issue to added a PRs ready for review category.  Also tried to reinstate PRs Complete category.

mrmcwethy, 2017-12-12:
I as working on Char_LCD and RGB_Display

mrmcwethy, 2017-12-13:
I am starting on pylint of SD

mrmcwethy, 2017-12-13:
I am picking up SimpleIO.

mrmcwethy, 2017-12-14:
i also did wavefrom

tannewt, 2018-01-09:
I'm doing IS31FL3731. I believe its the last one.

mrmcwethy, 2018-01-23:
There was a lot of work involved with incorporating pylint into the library build process and it is complete. Thanks everyone!

closed 2018-01-23 by mrmcwethy