adafruit/circuitpython #478: Support of the Feather M0 WiFi ATSAMD21 + ATWINC1500

IeiuniumLux, 2017-12-12:
Circuitpython listed Arduino Zero as a supported board, can it also be installed on the Feather M0 WiFi ATSAMD21 + ATWINC1500 (product ID: 3010)?

https://www.adafruit.com/product/3010

jerryneedell, 2017-12-12:
Yes, but it will be treated as a Feather M0 Basic. There is no support for the Wifi at this time.

IeiuniumLux, 2017-12-12:
OpenMV also uses the [WINC](https://openmv.io/collections/shields/products/wifi-shield) chip, will it possible to contribute that [code](https://github.com/openmv/openmv/blob/9cc209d128fd9c7a6650c53e34816c8940af0f5b/src/omv/py/py_winc.c) to Circuitpython?

closed 2017-12-12 by IeiuniumLux

tannewt, 2017-12-12:
@IeiuniumLux it'd be possible to port but its not a focus of ours. Let me know if you are interested in trying to port it and I can give pointers on how our structure is different. Thanks!

IeiuniumLux, 2017-12-12:
Sure, I can give it a try since I have one of the Feather M0 with the ATWINC1500 WiFi module.  I'll send send a push request when I have something for you to review.

tannewt, 2017-12-12:
@IeiuniumLux Awesome! Feel free to send it early to make sure you are on the right track. I'm regularly on our Discord chat too if you have questions. Link to join is: https://adafru.it/discord