adafruit/circuitpython #487: DMA'd ADC for high-speed sampling -> buffer

ladyada, 2017-12-15:
for some applications, we want to grab a chunk of samples from ADC at 'fast' (16KHz!) fixed periods. lets think of a way we can request an array of 16-bit **value**s that will be popped off the ADC pin at **samplerate**

tannewt, 2017-12-15:
Perhaps this would be `AudioIn`? To correspond with [AudioOut](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/audioio/AudioOut.html)

tdicola, 2017-12-15:
This would be used for lots of things beyond audio though--sampling any signal from the ADC, like a piezo, light sensor, load cell, etc.

ladyada, 2017-12-15:
Yah - i think its the mirror. _technically_ its not just for Audio, and i dont think we'd do WAV translation? but maybe we would?

ladyada, 2020-02-05:
@jepler this might be a good one to pick up soonish (since ulab would be a good pairint)

kevinjwalters, 2020-04-20:
I'm interested in this too. Here's some previous forums discussions where this has come up:

- [ADC / DAC very slow](https://forums.adafruit.com/viewtopic.php?f=60&t=152195) (on a Feather M4 express)
- [Is there a blocking analog read and what is sample rate?](https://forums.adafruit.com/viewtopic.php?f=60&t=142006)

I'm not sure if it's related but I've noticed some demand for performant circular buffers discussed in [CircularBuffer equivalent in CircuitPython?](https://forums.adafruit.com/viewtopic.php?f=60&t=164138).

It would be great to be able to choose the 14bit ADC mode on the nRF52840. I'm curious if that gives better results than doing it in software. It's certainly more efficient in terms of CPU cycles.

dhalbert, 2020-04-20:
I would like to add more control over the ADC, such as specifying oversampling, averaging (if available), maybe even differential mode, etc.

The native nRF52840 ADC accuracy is 12 bits. You can set it to 14 bits, but that just shifts the result over if oversampling is not enabled. But with oversampling, you can get 14 real bits.

kevinjwalters, 2020-04-20:
I've just put some graphs of the nRF52840's 12 bit mode in action in CircuitPython 5.2.0 in [Changing ADC to over-sampling mode for higher resolution?](https://forums.adafruit.com/viewtopic.php?f=60&t=164636).

kevinjwalters, 2020-04-23:
Added some performance data on a few different ways to read and sometimes store samples in https://forums.adafruit.com/viewtopic.php?f=60&t=164758 - nothing better than 21-21kHz on a CLUE.

latkinso42, 2022-08-23:
Please review https://github.com/latkinso42/circuitpython branch adcdma as a possible solution or possible solution to this issue. Thanks!