adafruit/circuitpython #488: Ability to change native resolution of DAC and ADC's

ladyada, 2017-12-15:
it would still be padded to 16-bits but perhaps a **.resolution** setter? see similar  https://www.arduino.cc/reference/en/language/functions/zero-due-mkr-family/analogreadresolution/

Low priority!

tdicola, 2017-12-15:
I'd definitely add a getter to read resolution, maybe even in 2.2?  This would get people into a mode of using the resolution as read from the hardware in their code early.  It would allow changing the resolution easily in the future (as people wouldn't bake in the 65535, etc. limits like we do today).

ladyada, 2017-12-15:
we'd still pad to 16-bits so there's no change in API! resolution changes the precision of that 16 bits only

tdicola, 2017-12-15:
Ah gotcha, one thing it's pretty uncommon to have a write-only property in python.  Typically there's always a getter to allow you to see what state the property is in without keeping track of it yourself. 
 Would the getter for this resolution prop always return 16-bit though?  I.e. I set to 12-bit, 8-bit, etc. but always read back 16-bit since it's internally scaling?

dhalbert, 2017-12-15:
I've targeted 2.2 as mostly fixes; 3.0 or later makes more sense for API changes/enhancements

ladyada, 2017-12-15:
edited to change it to 'property' rather than setter