adafruit/circuitpython #493: All bundle library should have an examples folder

mrmcwethy, 2017-12-20:
Here is the list of libraries that still need an examples folder:

- [ ] Adafruit_CircuitPython_RTTTL
- [ ] Adafruit_CircuitPython_SD
- [ ] Adafruit_CircuitPython_IS31FL37
- [ ] Adafruit_CircuitPython_Thermistor
- [ ] Adafruit_CircuitPython_BNO055
- [ ] Adafruit_CircuitPython_DotStar
- [ ] Adafruit_CircuitPython_DS1307
- [ ] Adafruit_CircuitPython_MCP98
- [ ] Adafruit_CircuitPython_MAX7219
- [ ] Adafruit_CircuitPython_ADS1x15
- [ ] Adafruit_CircuitPython_PCF8523
- [ ] Adafruit_CircuitPython_HID
- [ ] Adafruit_CircuitPython_DS18X20
- [ ] Adafruit_CircuitPython_Waveform
- [ ] Adafruit_CircuitPython_SimpleIO
- [ ] Adafruit_CircuitPython_OneWire
- [ ] Adafruit_CircuitPython_NeoPixel
- [ ] Adafruit_CircuitPython_TSL2561
- [ ] Adafruit_CircuitPython_CircuitPlayground
- [ ] Adafruit_CircuitPython_Register
- [ ] Adafruit_CircuitPython_BusDevice
- [ ] Adafruit_CircuitPython_MAX31855

Sigafoos, 2018-01-04:
I'm not seeing examples for some of the repos that you have checked off: Adafruit_CircuitPython_SD for example.

mrmcwethy, 2018-01-04:
@Sigafoos The checked ones are ones that I plan on adding examples to.  If you want to help with examples, pick one that is not check yet.

Sigafoos, 2018-01-04:
@mrmcwethy got it!

mrmcwethy, 2018-01-04:
@Sigafoos BTW: if you really like any of the ones I selected, please let me know, you can do em.

Sigafoos, 2018-01-06:
I believe SimpleIO can be closed after adafruit/Adafruit_CircuitPython_SimpleIO@6667bb6

mrmcwethy, 2018-01-06:
Ok, I moved SimpleIO to the Completed PR list.

mrmcwethy, 2018-01-09:
https://github.com/adafruit/Adafruit_CircuitPython_MAX7219
https://github.com/adafruit/Adafruit_CircuitPython_BusDevice
https://github.com/adafruit/Adafruit_CircuitPython_HID

Ready for review.

mrmcwethy, 2018-01-16:
@tannewt to get examples done for the current library, i had to lint it as well. These are examples that will need to be rewritten for you experimental version of IS31FL3731.  Let me know if you want me to finish that work.

tannewt, 2018-01-16:
@mrmcwethy I'll merge it now.

I'm actually tempted to punt on the experimental version for a bit longer until we know how we want all of the displays to work.

mrmcwethy, 2018-02-09:
We finally have examples folders for all the Adafruit Bundle Libraries.  Thanks everyone!

closed 2018-02-09 by mrmcwethy

ladyada, 2018-02-09:
AMAZING!!!! thanks x 1000 for your help @mrmcwethy

tannewt, 2018-02-09:
🎉

dhalbert, 2018-02-09:
@mrmcwethy thank you and thank everyone who else who contributed examples! The libraries are in great shape!