adafruit/circuitpython #501: file.seek

Kabaju42, 2017-12-26:
file.seek() does not work from the current position. It does work for seeking from the beginning or end. Example from the REPL:

Adafruit CircuitPython 2.1.0 on 2017-10-17; Adafruit CircuitPlayground Express with samd21g18
>>> f = open('songnight1.mid', 'rb')
>>> f.seek(2)
2
>>> f.seek(4, 0)
4
>>> f.seek(-2, 1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 95
>>> f.seek(-40, 2)
2116

dhalbert, 2017-12-26:
The second arg ==1 is deliberately not implemented. Not sure why:
https://github.com/adafruit/circuitpython/blob/36ec29d4e85b19487c868287f09d2492eecc0274/extmod/vfs_fat_file.c#L129
It seems to me it could call `f_tell()` and then add that value to 
`s_offset` and call `f_lseek()`. @tannewt, do you know it might not be implemented? I looked in the MicroPython forum and issues and couldn't find a mention of this. I wrote an issue to ask: https://github.com/micropython/micropython/issues/3517.
```
    if (request == MP_STREAM_SEEK) {
        struct mp_stream_seek_t *s = (struct mp_stream_seek_t*)(uintptr_t)arg;

        switch (s->whence) {
            case 0: // SEEK_SET
                f_lseek(&self->fp, s->offset);
                break;

            case 1: // SEEK_CUR
                if (s->offset != 0) {
                    *errcode = MP_EOPNOTSUPP;
                    return MP_STREAM_ERROR;
                }
                // no-operation
                break;

            case 2: // SEEK_END
                f_lseek(&self->fp, f_size(&self->fp) + s->offset);
                break;
        }

        s->offset = f_tell(&self->fp);
        return 0;

    } else if (request == MP_STREAM_FLUSH) {
        FRESULT res = f_sync(&self->fp);
        if (res != FR_OK) {
            *errcode = fresult_to_errno_table[res];
            return MP_STREAM_ERROR;
        }
        return 0;

    } else {
        *errcode = MP_EINVAL;
        return MP_STREAM_ERROR;
    }
}
```

tannewt, 2017-12-26:
I have no idea. Should we add this to the 3.0 milestone as polish?

dhalbert, 2017-12-26:
I'll see what the MPy folks, say, if anything, but yeah, it's minor.

dhalbert, 2018-01-09:
Posted an issue here https://github.com/micropython/micropython/issues/3517, and also see this PR: https://github.com/micropython/micropython/pull/3494

PR #697 (2018-03-23): extmod/vfs_fat_file: Implement SEEK_CUR for non-zero offset.
I noticed that this functionality was now added in uPy and cherry-picked the fix.

As far as testing goes, there is an added test case but unfortunately it looks like the related test is skipped during the unix and qemu builds. ☹️ 

Closes: #501
closed 2018-03-23 by tannewt