adafruit/circuitpython #504: can't search for code as long as this repo is a fork

boneskull, 2017-12-30:
GitHub's forking functionality is implemented with non-permanent forks in mind.  Because of this, GitHub has *limited* forked repositories in several ways; one of those limitations is that searching for code within them is *completely disabled.*

CircuitPython's codebase is different enough from MicroPython's that I cannot search for what I want to find within that repo, as is suggested by the UI:

[image]

I think all you need to do is:

1. Delete this repository (scary, I know)
1. Create it again as a new, empty repo 
1. Push a current working copy.  

That should break the "fork" link (unless there's a better way to do it).  This does not impact your ability to pull code from MicroPython.

It's clear from the `README` that CircuitPython is a fork of MicroPython, in case anyone was concerned.  :wink:

deshipu, 2017-12-30:
Or you can work around this yourself: clone this repository, create a new repository on Github under your own user, push the clone into it, search there.

dhalbert, 2017-12-30:
CIrcuitPython is a fork, but is not a detached fork. We still pull and merge from MicroPython on a regular basis, so we don't want to detach from that repo.

A number of us do local searches in our clones using the nice tool The Silver Searcher: https://geoff.greer.fm/ag/ and https://github.com/ggreer/the_silver_searcher.

closed 2017-12-30 by dhalbert

boneskull, 2017-12-30:
cloning the repo to simply search isn’t particularly user-friendly. I’m not sure I understand why this repo needs to be “attached” to MicroPython, as it’s trivial to pull code from any remote via `git`.

tannewt, 2018-01-02:
Hopefully GitHub will add fork search in the near future. I agree it'd be good to have.

I want to leave us as a fork because its one way we can be clear that we're based on MicroPython. We already get criticized for forking. I don't want to hide it.

FutureSharks, 2019-03-22:
The lack of search is really annoying! I just want to find the code for `rotaryio`: https://github.com/adafruit/circuitpython/search?q=forked&type=Code

:(

@dhalbert not being a fork doesn't affect the ability to work with the main micropython repo. You can still pull and merge between repos that aren't forked.

tannewt, 2019-03-26:
@FutureSharks Our code for modules is consistently structured so it should be easy to find. The layout is described here: https://circuitpython.readthedocs.io/en/latest/docs/common_hal.html

Breaking the fork connection on GitHub will make it less obvious that we are a fork of MicroPython. I want to leave it as a way to make our relationship to the original project clearer.

FutureSharks, 2019-03-27:
OK fair enough, thanks for the reply @tannewt