adafruit/circuitpython #511: Implement I2C clock stretching in bitbangio

jerryneedell, 2018-01-04:
Some I2C sensors require clock stretching and do not work properly when using bitbangio (necessary on the esp8266).  Examples are CCS811 https://github.com/adafruit/Adafruit_CircuitPython_CCS811/issues/9  and bno055.

deshipu, 2018-01-04:
This is strange, because I remember that I implemented clock stretching for MicroPython for the software I2C... Could it be that CircuitPython didn't use that?

Looking at the code, it's there: https://github.com/adafruit/circuitpython/blob/master/shared-module/bitbangio/I2C.c#L53

Do you have a logic analyzer to capture what is happening with that sensor?

jerryneedell, 2018-01-04:
I do have some logic analyzer captures - GitHub won't accept the file types. I can get them to you in discord.

jerryneedell, 2018-01-04:
I'll have to regenerate them, but it won't be today -  the ones I have are too large because I captures some analog data as well. I'll try to repeat the tests tomorrow and post the files.  Sorry for the confusion.

jerryneedell, 2018-01-04:
here is a screenshot of the Logic analyzer capture - perhaps it will tell you enough
[image]

jerryneedell, 2018-01-04:
this is the start of the capture 
[image]

jerryneedell, 2018-01-05:
I lowered the I2C frequency to 10000 and it works better, but still occasionally fails withe either an OSError 5 as above or sometimes a RunTime error 2. I don’t have a logic analyzer capture of it at 10000. Would you like to see one?

deshipu, 2018-01-05:
What value of pull-up resistors are you using? Do you have analog pins on your logic analyzer, or an oscilloscope, so that you can actually see the shape of the signal? How long are the wires?

jerryneedell, 2018-01-05:
The breakout board has 10k pulllups. I’ll try to get some scopeshots to post this morning. I was having some trouble getting good shots yesterday. I also have another sensor, mcp9808 also with 10k pull-ups connected. Wires are about 15 cm. to ccs811.

jerryneedell, 2018-01-05:
ccs811 - 10Khz clock
[image]

ccs811 - default clock
[image]
ccs811 - default clock
[image]

deshipu, 2018-01-05:
Those signals look reasonable, I don't think this is the problem.

jerryneedell, 2018-01-05:
I have tried several approaches- here is a summary:
slow clock rate to 10Hkz -- works "better" runs for 10s of seconds or minutes then fails

increase Clock stretching limit to 512 - works a bit better -sometimes works at default clock rate - still fails eventually even at 10Khz

noted that 3V line was sagging (sometimes below 3V) when ccs811 running. powered with 5Vin and level shifters on SCL/SDA
still tight increased clock stretch timeout  it works "better" It now runs at both default clock rate and 10Khz, but still crashes after 10s of seconds to minutes.  3.3v line is better but still sags to < 3.1 V.

running out of ideas.

jerryneedell, 2018-01-05:
After much discussion with @deshipu and many tests on esp8266 and feather m0 (bit banging) there does not appear to be any problem with i2c clock stretching. At least the issues with the ccs811 do not appear to be related to clock stretching so it is time to close this issue.

closed 2018-01-05 by jerryneedell