adafruit/circuitpython #514: Dostar not working uder 3.0

jerryneedell, 2018-01-08:
I am unable to get a simple dotstar test working on a Trinket_m0 under 3.0.
This works under 2.x

I have placed the 3.x adafruit_dotstart.mpy in /lib.

`
Adafruit CircuitPython 3.0.0-alpha.1-51-g1805bca-dirty on 2018-01-08; Adafruit Trinket M0 with samd21e18
>>> 
>>> 
>>> from board import *
>>> import adafruit_dotstar as dotstar
>>> pixel = dotstar.DotStar(APA102_SCK, APA102_MOSI, 1, brightness=.1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_dotstar.py", line 70, in __init__
ValueError: Pin PA00 in use
>>> 

`

siddacious, 2018-02-22:
I've been looking into getting the Dotstar to work for my m4 board so I can take a look at this now that I have a trinket and bitsy.

jerryneedell, 2018-02-23:
FYI - Just tried with current master - issue is still present:
```
Adafruit CircuitPython 3.0.0-alpha.1-205-g6daf4bd on 2018-02-23; Adafruit Trinket M0 with samd21e18
>>> 
>>> 
>>> from board import *
>>> import time
>>> import adafruit_dotstar as dotstar
>>> pixel = dotstar.DotStar(APA102_SCK, APA102_MOSI, 1, brightness=.2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_dotstar.py", line 70, in __init__
ValueError: Pin PA00 in use
>>> 
```

siddacious, 2018-02-23:
Thanks for the update. The reason you're getting that specific error is that the PA00 and PA01 pins aren't being reserved for dotstar use because the status led code for dotstars is broken at the moment and I don't think it compiles with them enabled. 

I got it to compile but it was crashing somewhere in the SPI setup code so I'll have to take a closer look tonight or this weekend. I'll update with my progress as appropriate.

siddacious, 2018-02-24:
I was able to recompile after removing the PA00 and PA01 from the reserved pins and the above example works just fine. My trinket m0 mpconfigboard.h currently looks like
```
#define MICROPY_HW_BOARD_NAME "Adafruit Trinket M0"
#define MICROPY_HW_MCU_NAME "samd21e18"

// Rev B - Black
// #define MICROPY_HW_APA102_MOSI   (&pin_PA00)
// #define MICROPY_HW_APA102_SCK    (&pin_PA01)

#define MICROPY_PORT_A        (PORT_PA24 | PORT_PA25)
//#define MICROPY_PORT_A        (PORT_PA00 | PORT_PA01 | PORT_PA24 | PORT_PA25)
#define MICROPY_PORT_B        (0)
#define MICROPY_PORT_C        (0)

#include "internal_flash.h"

#define CIRCUITPY_INTERNAL_NVM_SIZE 0

#define BOARD_FLASH_SIZE (0x00040000 - 0x2000 - 0x010000)
```
Since the `MICROPY_HW_APA102_MOSI` defines are still commented out, the status dotstar still doesn't work. I'll try putting them back in but unless something has changed since I last tested it won't compile.

(1 minute later):
Indeed it doesn't compile with the same error that I saw previously:
```
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
../../supervisor/shared/rgb_led_status.c: In function 'rgb_led_status_init':
../../supervisor/shared/rgb_led_status.c:73:26: error: 'busio_spi_obj_t {aka struct <anonymous>}' has no member named 'current_baudrate'
         if (status_apa102.current_baudrate > 0) {
                          ^
../../supervisor/shared/rgb_led_status.c:77:13: error: implicit declaration of function 'spi_disable' [-Werror=implicit-function-declaration]
             spi_disable(&status_apa102.spi_master_instance);
             ^~~~~~~~~~~
../../supervisor/shared/rgb_led_status.c:77:13: error: nested extern declaration of 'spi_disable' [-Werror=nested-externs]
../../supervisor/shared/rgb_led_status.c:77:39: error: 'busio_spi_obj_t {aka struct <anonymous>}' has no member named 'spi_master_instance'
             spi_disable(&status_apa102.spi_master_instance);
```

dhalbert, 2018-02-24:
The baudrate issue is on my list to fix.

dhalbert, 2018-02-24:
Oh, wait that's a different baudrate issue, but still, can look at it.

siddacious, 2018-02-27:
A quick update: I was able to get the dotstar working on my trinket m0 though testing on my m4 board I get an OSError 5 (IO Error I think) and on my itsy bitsy I don't get MSC or CDC anymore; I'll dig more tonight.

siddacious, 2018-03-08:
Ok, I think I finally got it working; I was able to get the status dotstar status-ing and then was able to re-purpose the same sercom for a SPI connection. I tested on my m4 board so I'll test on the trinket, bitsy, m4 metro and perhaps others before submitting a PR tonight.

tannewt, 2018-03-09:
Fixed by #662

closed 2018-03-09 by tannewt