adafruit/circuitpython #523: Add Driver for the Adafruit TSL2591 High Dynamic Range Digital Light Sensor

Apotessar, 2018-01-12:
Hi

It would be cool if circuitpython would support the Adafruit TSL2591 High Dynamic Range Digital Light Sensor right out of the box (or within the bundle). Seems legit to me since this is the most capable light sensor in the Adafruit portfolio, and the only one with is able to handle the range needed for modern photography (as support).
Actually I will work on my own driver, witch i'am willing to share when it is done, but first I need to learn more about i2c communication and python in genrell.

Thanks and keep up the good work (finaly I don't have to deal with C anymore ;-) )
Regards
Pascal

ladyada, 2018-01-16:
ok we'll take a look and close this issue when its added :)

caternuson, 2018-01-24:
OK to close?
https://github.com/adafruit/Adafruit_CircuitPython_TSL2591

ladyada, 2018-01-24:
yah lets close - we have a driver :)

closed 2018-01-24 by ladyada