adafruit/circuitpython #526: nRF Port: Get Remaining Shared Modules Working

sommersoft, 2018-01-15:
The shared modules below are not currently included in the nRF port build (taken from Makefile). Obviously, some will be easier than others so I put them in as a task list.

- [ ] bitbangio/__init__.c
- [ ] bitbangio/I2C.c
- [ ] bitbangio/OneWire.c
- [ ] bitbangio/SPI.c
- [ ] busio/OneWire.c
- [ ] gamepad/__init__.c
- [ ] gamepad/GamePad.c
- [ ] struct/__init__.c
- [ ] uheap/__init__.c
- [ ] ustack/__init__.c

microbuilder, 2018-01-15:
Thanks for putting a list together!

arturo182, 2018-02-03:
I added gamepad with a example in https://github.com/arturo182/circuitpython/commit/823940ef2001a46eaff3a9bdb4c29cc3cae91bec but it is on top of my pca10056 changes and seems github doesn't allow me to push it into a new PR that depends on #573 and I don't want to mix those. So I can wait for this one to be merged before pushing a  new PR unless there is some way to do it in a clean way in Github (sorry I come from Gerrit world).

arturo182, 2018-02-03:
Added `struct` in https://github.com/arturo182/circuitpython/commit/71227c5b34f5658a3daadbf3e15ffa3329e417bb, same story as above :)

tannewt, 2018-02-04:
@arturo182 you can do separate PRs if you start new branches locally. Star them based on adafruit's master and they can be reviewed in parallel.

arturo182, 2018-02-04:
Yes I am aware of that but it seems Github doesn't support PRs that are on top of other PRs, so I could commit a series of commits and put them into separate commits because the PR will always include all commits not in the main branch. For example the LED fix depends on the nRF52840 PR so I wish i could make a LED PR which is marked to depend on the nRF one and can't be merged until that one is. But I can see how that could get complicated and confusing fast ;)

tannewt, 2018-02-05:
Are the shared commits required for your dev setup?  You could use rebase to remove the shared commits if not. Waiting is ok too. I'll do the review Tuesday when I have time if @microbuilder doesn't get to it.

arturo182, 2018-02-05:
I do some changes in these commits to the pca10056 files that i added in the other PR so I think I'll wait, no hurry TBH ;)

arturo182, 2018-02-06:
If I understand correctly the `audiobusio` module offers a interface for I2S which the nRF52 family has so that module could also be implemented in the nRF port! :)

dhalbert, 2018-09-17:
Broke out remaining tasks into individual issues and closed in favor of them. (e.g. #1194).

closed 2018-09-17 by dhalbert