adafruit/circuitpython #530: Deal with multiple boards showing same drive name on Mac OSX (and maybe other OS's)

tdicola, 2018-01-16:
I noticed while doing some radio library development that I needed to have two CircuitPython boards (M0-based) connected to my Mac at once (one for transmitting, the other for receiving).  This presented an interesting problem as both boards show up in Finder with the exact same drive name (CIRCUITPY):
[image]
If you use the command line you can see a small difference in the /Volumes directory with the name (one is given /Volumes/CIRCUITPY/ and another /Volumes/CIRCUITPY 1/):
[image]
In both cases though it's really hard to figure out which board to copy a file to as they both appear the same.  A quick workaround is to add a dummy text file to the board so you can see it in finder and realize the difference.  However I think we should consider some way to disambiguate the board names.  This would greatly help scenarios like a classroom where a teacher might have multiple boards connected to set them up, etc.  Or just normal users with two boards connected to their computer.

Some thoughts mentioned in discussion:
- Generate the filesystem label/name with a hash or few bits from the board's unique ID at the end.  So CIRCUITPY could be CIRCUITPY_1AF etc.
- Allow users to change the name of the filesystem label programmatically, like in boot.py.  This is actually kind of interesting because it would allow very clear board-specific names.  Like a temperature sensor node could be named TEMP1, TEMP2, and be very easy to spot when connected to a computer.

tannewt, 2018-03-23:
@jepler I agree with you that the user could do it as needed. A lot of our guides reference CIRCUITPY and it'd be good to be exact by default.

How about we add something to `storage` that can change it during `boot.py`? That way the process is OS agnostic but still done by the user.

jepler, 2018-03-23:
@tannewt interesting idea, I will try to take a look at doing that this weekend.  I am not familiar with the storage module yet, but maybe the API should look something like: `storage.setlabel(filesystem, "newlabel")` ?  I'm not actually seeing in the docs what sort of values you might pass for `filesystem` though, hm.

tannewt, 2018-03-23:
IIRC `remount` uses the mountpoint. So `"/"` or `"/sd"`.

jepler, 2018-03-31:
@tannewt with #712 merged, can we close this up?  or is there more to do (tutorial documentation to be written?)

tannewt, 2018-04-01:
Closing sounds good! I'll try and remember to highlight its addition in the next set of release notes. Thanks!

closed 2018-04-01 by tannewt