adafruit/circuitpython #531: Compress QSTRs and raw strings

tannewt, 2018-01-16:
The contents of QSTRs can be quite repetitive. Perhaps a dictionary compression scheme would use less memory to store the strings.

[image]

dhalbert, 2018-06-01:
I did a rough count of how many characters are used for exception strings:
```
g -l 'raise.*"' |egrep 'ports/atmel-samd|py/|shared-' | xargs egrep -h 'raise.*"' | grep -v // | sed -E 's/.*"(.*)".*/\1/' | sort -u | wc
    264    1293    7727
```
If I run `compress` (uses LZW) on the strings, it shrinks from 7727 bytes to 4183.

ladyada, 2018-11-19:
@tannewt i think u can close this right?

tannewt, 2018-11-19:
We currently only compress static string literals (like exception messages). We don't actually do QSTRs or dynamic strings yet. So, there is something we can do for this still.

dhalbert, 2019-03-26:
@tannewt The huffman coding is not dictionary-based across strings, is that right? I looked for a library that does dictionary-based compression across multiple strings with a shared dictionary, but didn't find anything. All the libraries seem to be very stream oriented. I'd think we might do well with a dictionary scheme because the messages have quite a few substrings in common.

tannewt, 2019-03-26:
@dhalbert Correct. It's only individual byte frequency.

ngrams could help for unicode characters and common substrings but adds overhead in the encoding of the huffman tree itself.

Another thing we could do is apply the huffman compression to QSTRs. It's more challenging because QSTRs are dynamically added as well as static. We'd save memory by doing of for string objects as well.

dhalbert, 2026-09-15:
Claude wrote this:

Closing this after measuring what compressing qstr text would actually buy, using the SAMD builds as of 2026-09-15.

Theoretical savings on the const qstr text (the strings only, excluding pointers and lengths):

| Board | qstrs | text bytes | zlib | order-0 Huffman (the translation scheme) |
|---|---|---|---|---|
| trinket_m0 | 638 | 5602 | 2747 | ~3450 |
| metro_m0_express | 941 | 8839 | 4157 | ~5480 |
| metro_m4_express | 1651 | 15756 | 7012 | ~9860 |

So the ceiling is roughly 1.7 KB on a samd21 non-express board and 4.4 KB on samd51 with the Huffman coder already in the tree, and it is not reachable in practice:

- `qstr_str()` and `qstr_data()` return pointers into ROM that callers keep indefinitely. Compressed qstrs would need decompression on every access plus a buffer-lifetime scheme, which touches string handling throughout `py/`. Translation compression only works because error text is consumed immediately by `mp_decompress_rom_string()`.
- `qstr_find_strn()` compares raw pool bytes on every identifier intern during compile and import. Comparing against compressed entries means decompressing candidates in the binary search, which slows down exactly the boards that need the space.
- The one cheap piece, sharing common suffixes, the linker already does through string merging: 429 bytes on trinket_m0 and 1283 bytes on metro_m4_express.

The per-qstr overhead rather than the text is the better target; see #5276.

closed 2026-09-15 by dhalbert