adafruit/circuitpython #537: Need Feather M0 RFM69 build with digital pins D3, D4, D8 exposed.

tdicola, 2018-01-18:
We need a new build of both 2.2 and later 3.0 which is for the feather M0 RFM69 board (https://learn.adafruit.com/adafruit-feather-m0-radio-with-rfm69-packet-radio?view=all#downloads).  Unfortunately our basic M0 build only exposes the pins on the outside of the feather and is unable to access the CS, RST, and interrupt lines of the RFM radio built-in to this feather (which use pins routed internally).  Here's the mapping to SAMD21 pins:

D3 = RFM69 D0/interrupt = SAMD21 PA09 (this line is currently unused by the RFM69 library but we should expose it for the future)
D4 = RFM69 RST = SAMD21 PA08
D8 = RFM69 CS = SAMD21 PA06

Please create a Feather M0 RFM69 build that includes at least these pins as new digital inputs/outputs routed to the appropriate SAMD21 pins.  If you need a reference the schematic is here: https://cdn-learn.adafruit.com/assets/assets/000/032/914/original/feather_schem.png?1465421956

Alternatively I'd recommend we add to our basic build or also create a new 'super' M0 build that exposes _all_ SAMD21 pins.  This would save some pain in the future if people are using new boards, other hardware, etc. before custom builds are ready.

tannewt, 2018-01-18:
I'm trying to keep 2.x very limited going forwards so we can focus on 3.x.

All SAMD21 pins are exposed in `microcontroller.pin`:

```
>>> from microcontroller import pin
>>> dir(pin)
['PA00', 'PA01', 'PA02', 'PA03', 'PB08', 'PB09', 'PA04', 'PA05', 'PA06', 'PA07', 'PA08', 'PA09', 'PA10', 'PA11', 'PB10', 'PB11', 'PA12', 'PA13', 'PA14', 'PA15', 'PA16', 'PA17', 'PA18', 'PA19', 'PA20', 'PA21', 'PA22', 'PA23', 'PA24', 'PA25', 'PB22', 'PB23', 'PA27', 'PA28', 'PA30', 'PA31', 'PB02', 'PB03']
>>>
```

For 3.x I think its fine to have specific builds so that its clear to users what they need.

tdicola, 2018-01-18:
I'd recommend taking this for the simplicity of users who want to use the Feather M0 RFM69.  The alternative is loading the M0 feather basic build (not obvious or mentioned in docs) and using the alternative pins from microcontroller.  We built the tooling to automatically generate builds so adding new ones has minimal overhead right?

If it is the case that adding new builds is too much work for intra-milestone releases I would go ahead and make builds for all the Feathers proactively like WINC1500, FONA, etc. to save any future pain and user confusion.

tannewt, 2018-01-18:
My goal is to focus us on 3.0 beta. I'm hoping that we can release our first beta in a few weeks. Adding more boards in 2.x is extra work because board definitions have changed with 3.x and ASF4.

I only want to modify 2.x if we need it for a hardware release like the ItsyBitsy.

tdicola, 2018-01-18:
So would you accept a pull for adding the board in 2.x then if it's too much work for others to do now?  Sorry I'm not following the need for extra work.  From the user perspective this is new circuitpython support for hardware that's been requested a lot.  I'd rather support it well out of the gate vs. hold on it longer to an unknown time with 3.x.

tannewt, 2018-01-18:
If you want to do it, then please do it in both 2.x and master.

tdicola, 2018-01-18:
Sounds good, merge these requests to add the missed boards:
2.x: https://github.com/adafruit/circuitpython/pull/538
master: https://github.com/adafruit/circuitpython/pull/539
I added the RFM69, RFM9x, nRF51, and WINC1500 to cover all of the Feather M0 boards.  They're based on the Feather M0 basic but add pin definitions for the internally routed peripheral control lines. This will make it much easier for users to use these feathers, and help anyone who wants to start on support for WINC1500, RFM9x, etc. with a feather.

ladyada, 2018-01-24:
done in 2.2.1 release - closing!

closed 2018-01-24 by ladyada