adafruit/circuitpython #542: Recursion error on ESP8266

jerryneedell, 2018-01-20:
WHn I run this code on an ESP8266 it fails as shown below.  It runs successfully on an M0 (feather_m0_adalogger)

```python
import adafruit_sdcard
import busio
import digitalio
import board
import storage

# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(board.GPIO2)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

# Use the filesystem as normal.
with open("/sd/test.txt", "r") as f:
    print(f.readline())
```
runs on M0 - fails on ESP8266

```Adafruit CircuitPython 2.2.0-5-g544b9e4 on 2018-01-20; ESP module with ESP8266
 import sdread
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdread.py", line 12, in <module>
  File "adafruit_sdcard.py", line 360, in readblocks
  File "adafruit_sdcard.py", line 238, in _block_cmd
  File "adafruit_sdcard.py", line 207, in _cmd
RuntimeError: maximum recursion depth exceeded
```

Note:If I change the last lines to the following - it works on the ESP8266
```
f=open("/sd/test.txt", "r") 
print(f.readline())
f.close
```

jerryneedell, 2018-05-17:
FYI -Just verified that this issue is still present.

tannewt, 2018-06-12:
I'll look at this now. I don't think its sdcard specific and is just a matter of the stack protection settings that lead to the maxrecursion error.

tannewt, 2018-06-12:
This is an issue in 2.3.1 as well.

The ESP8266 memory map is very complicated and the stack space is shared with the underlying RTOS that handles the WiFi side of things. So, I don't think we have any options here. If we increase the limit on stack size we risk starving the RTOS of stack space. We could dive into micropython's stack allocations but that's complicated.

So, I'm closing this. Please use a different board for SD card support.

Background post about ESP8266 memory is here: https://www.kickstarter.com/projects/214379695/micropython-on-the-esp8266-beautifully-easy-iot/posts/1501224

closed 2018-06-12 by tannewt