adafruit/circuitpython #543: ESP8266 build fails for 3.0

jerryneedell, 2018-01-22:
I am unable to build the esp8266 image under 3.0 master  - the following error occurs
at the end of the build:

```
GEN build/genhdr/qstrdefs.generated.h
FREEZE modules
common-hal/microcontroller/Processor.c: In function 'common_hal_mcu_processor_get_uid':
common-hal/microcontroller/Processor.c:43:5: error: implicit declaration of function 'system_get_chip_id' [-Werror=implicit-function-declaration]
     uint32_t id = system_get_chip_id();
     ^
cc1: all warnings being treated as errors
../../py/mkrules.mk:53: recipe for target 'build/common-hal/microcontroller/Processor.o' failed
make: *** [build/common-hal/microcontroller/Processor.o] Error 1
```

It builds successfully for 2.x

jerryneedell, 2018-01-22:
It would appear that esp8266/hal-common/Processor.c is missing somthing:
```
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/circuitpython_master/ports/esp8266/common-hal/microcontroller$ diff Processor.c ~/projects/adafruit_github/circuitpython_2x/esp8266/common-hal/microcontroller/
41,47d40
< 
< void common_hal_mcu_processor_get_uid(uint8_t raw_id[]) {
<     uint32_t id = system_get_chip_id();
<     for (int i=0; i<4; i++){
<         raw_id[i] = id >> (i * 8);
<     }
< }
```

jerryneedell, 2018-01-22:
Sorry _ I now see that the update to add the UID was only applied to master, not to 2_x so it just appears that there is some error in the esp8266 configuration.

sommersoft, 2018-01-22:
I can try and duplicate later. But, yeah, the `UID` is only in master..def not in 2.x or any of the 3.0 tags.

@tannewt & @DanHalbert, I can pull the master over to 2.x & 3.0 on my fork and submit a PR...just let me know.

jerryneedell, 2018-01-22:
It looks like Travis did not build an esp8266 image when this was merged . Is that intentional?

jerryneedell, 2018-01-22:
looking for the function system_get_chip_id - grep does not find it anywhere in the source tree.
```
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/circuitpython_master$ grep -r "system_get_chip_id" *
ports/esp8266/modmachine.c:    uint32_t id = system_get_chip_id();
ports/esp8266/common-hal/microcontroller/Processor.c:    uint32_t id = system_get_chip_id();
Binary file ports/esp8266/build/modmachine.o matches
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:    uint32_t id = system_get_chip_id();
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:    uint32_t id = system_get_chip_id();
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);
ports/esp8266/build/genhdr/qstr.i.last:uint32 system_get_chip_id(void);

```

sommersoft, 2018-01-22:
`system_get_chip_id` is actually a native Espressif SDK instruction. it is also (and originally) used in `modmachine.c` for the `machine.unique_id` function. `modmachine.c` is in ports/esp8266 folder.

jerryneedell, 2018-01-22:
That what I thought. Since it looks likes some issue with the previous merge, I think it best for you to look it over and make the correction. I'm in no hurry. Let me know if you want me to do anything.

jerryneedell, 2018-01-22:
FYI - I made the change to add the include and build an image for some other testing. Just for fun, I thought I'd check the UID and this is what I get:
```>>> import microcontroller
>>> dir(microcontroller)
['__name__', 'cpu', 'delay_us', 'disable_interrupts', 'enable_interrupts', 'on_next_reset', 'reset', 'nvm', 'RunMode', 'Pin', 'pin', 'Processor']
>>> dir(microcontroller.cpu)
['frequency', 'temperature', 'uid']
>>> microcontroller.cpu.uid
bytearray(b'\xe8\x0cL\x00')
>>> microcontroller.cpu.frequency
80000000
```
Does that make any sense? what is 0x0cL?

sommersoft, 2018-01-22:
yeah, that's a correct result. for a human-readable format, import `struct` and print it using this: `struct.unpack('<i', microcontroller.cpu.uid)`

we kept the return type as a bytearray so that options were left open to how people wanted to use it.

jerryneedell, 2018-01-22:
hmm - whatever it is, it it the same as machine.unique_id()
```
Adafruit CircuitPython 3.0.0-alpha.1-65-g4bd1d09-dirty on 2018-01-22; ESP module with ESP8266
>>> 
>>> 
>>> 
>>> import microcontroller
>>> microcontroller.cpu.uid
bytearray(b'\xe8\x0cL\x00')
>>> import machine
>>> dir(machine)
['__name__', 'mem8', 'mem16', 'mem32', 'freq', 'reset', 'reset_cause', 'unique_id', 'idle', 'sleep', 'deepsleep', 'disable_irq', 'enable_irq', 'time_pulse_us', 'RTC', 'Timer', 'WDT', 'Pin', 'Signal', 'PWM', 'ADC', 'UART', 'I2C', 'SPI', 'DEEPSLEEP', 'PWRON_RESET', 'HARD_RESET', 'DEEPSLEEP_RESET', 'WDT_RESET', 'SOFT_RESET']
>>> machine.unique_id
<function>
>>> machine.unique_id()
b'\xe8\x0cL\x00'
>>> 
```

jerryneedell, 2018-01-22:
got it - thanks ```>>> import struct
>>> struct.unpack('<i', microcontroller.cpu.uid)
(4984040,)
>>> 
```

sommersoft, 2018-01-22:
yeah, it uses the exact same function. we added it to microcontroller to keep it standard with the other ports. on the esp8266, the "unique id" is the last 24bits of the MAC address. not quite as unique as other boards..but, its what we got. 😄

sommersoft, 2018-01-22:
i'm working on the fix right now...just have to make sure an rebase to avoid further damage. lol

sommersoft, 2018-01-23:
PR finally submitted. I have ruined my fork's history...to the ProGit book!

Thanks again @jerryneedell for catching this.

jerryneedell, 2018-01-24:
Fixed by #546

closed 2018-01-24 by jerryneedell