adafruit/circuitpython #545: add Driver fo SHT31D Temperature/Humidity Sensor

jerryneedell, 2018-01-22:
https://www.adafruit.com/product/2857

ladyada, 2018-01-25:
just adding a note, the arduino library is here https://github.com/adafruit/Adafruit_SHT31

deshipu, 2018-01-25:
And the MicroPython one is here: https://github.com/kfricke/micropython-sht31

jerryneedell, 2018-01-25:
Unless someone else want this one - you can assign it to me and I'll give it a shot.

ladyada, 2018-01-25:
goooo foooor iiiiiit :)

jerryneedell, 2018-01-28:
I have the basics working - can now read temperature and humidity and pass the checksums! The si7021 circuitpython library was a a great guide along with the Arduino  llibrary for the sht31. I hope to have something to share in the next few days.

jerryneedell, 2018-02-05:
Driver released

closed 2018-02-05 by jerryneedell

ladyada, 2018-02-06:
awesome work @jerryneedell - you did great!!! :)