adafruit/circuitpython #554: Document that sys.stdin.read() replaces Carriage Return characters with New line characters.

tballas, 2018-01-27:
This issue might serve as partial documentation.

This took a lot of digging to find. A carriage return sent over usb serial is replaced with a new line character when read with `e.g. sys.stdin.read(1).`

The source of the change seems to be the `stdio_read()` function in `\lib\utils\sys_stdio_mphal.c`

The reason for this behavior is unclear.

tannewt, 2018-01-31:
What does CPython do?

ladyada, 2021-02-20:
ditto on closing as wontfix, send people to dual CDC instead?

dhalbert, 2021-02-20:
We now have byte-based reading from the REPL stream or a second serial stream, unencumbered by REPL-specific characters: https://circuitpython.readthedocs.io/en/latest/shared-bindings/usb_cdc/index.html.

It does seem that the `\n` behavior is the same in CPython (same result on Windows and Linux):
```python
Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.stdin.read(2)
a
'a\n'
>>> 
```

closed 2021-02-20 by dhalbert