adafruit/circuitpython #561: nRF52840 port

arturo182, 2018-01-30:
What's the status of the nRF52840 port, is anyone working on it?

I followed the structure of the feather52 code and added the s140 6.0.0 softdevice to the download script and makefiles so it's used when building the nrf52840, but there is the case of adding support for it in the bluetooth driver.

If no one is currently working on adding s140 support in the bluetooth driver, I'd like to look into it. I have a lot of professional experience with the nRF52 series, so I believe it shouldn't be a very difficult task.

microbuilder, 2018-01-30:
It's on our radar, but no exact ETA since no modules are commercially available and the chips aren't in final production yet. Any PRs are always welcome for review, of course. We're currently looking at USB support on the nRF52840, but that's somewhat independent of the CP port itself.

arturo182, 2018-01-30:
Cool, then I guess I can have a stab at it, I have the nRF52840 DK and even a few nRF52840 chips, that I made into my own module (https://twitter.com/arturo182/status/951567482921332736) where I have USB working (and working on BLE). This is all in a private, hobby context.

tannewt, 2018-01-31:
@arturo182 what piece are you going to look at? I believe @hathach is looking into USB.

arturo182, 2018-01-31:
From what I've seen there is no support to even start building for the pca10056 so I started adding that and then also I added support for the v6.0.0 of the softdevice and now the build fails because Nordic modified the SDK between s132 and s140 so I wanted to get that working.

tannewt, 2018-01-31:
@arturo192 Ok perfect! Keep us posted if you have any questions! You can find me on [Discord](https://adafru.it/discord) most days.

arturo182, 2018-01-31:
Ok I got both peripheral (advertising, connecting, reading and writing), and scanner working on nrf52840 with SD 6.0.0, just need to clean up the code and can make a PR most likely Friday.

jerryneedell, 2018-02-01:
Wow! Sounds great! Looking forward to you PR. I’ll be happy to give it a try. Thank you.

microbuilder, 2018-02-01:
Sounds good, thanks. I'll double check the linker scripts, etc., to make sure they align with requirements for the internal filesystem, etc. I'm wrapping up another project today and tomorrow, but Friday is perfect for me to come back to this and have a look at what you've put together for a PR! We're poking at some higher level wrappers for BLE as well since the current API is pretty low level and not as friendly as it could be. If you have any example python files to push out to test against, though, I'm sure people would find that useful!

arturo182, 2018-02-01:
Yes, I added the flash fatfs in the linker scripts, for example i just used the peripheral and scanner in nrf/examples. I noticed the Central is commented out so I guess it's not available yet? I also have a question regarding the umachine module, I see it's missing in the module listbut some of the parts work, maybe we could re-add it? Is there any TODO/schedule/issue list for the nrf port so I could go through them and try fixing them?

tannewt, 2018-02-01:
@arturo182 The `machine` module is a MicroPython API so its not urgent for us to get working. If you'd like to then we'll merge it in. Our focus is the CircuitPython APIs defined in the `shared-bindings` and implemented in `common-hal` within each port. #526 has a list of some that are missing currently.

We have a short list of issues for nrf52 here: https://github.com/adafruit/circuitpython/issues?q=is%3Aissue+is%3Aopen+label%3Anrf52 Feel free to file more as you find them too.

microbuilder, 2018-02-01:
@arturo182 I'm currently looking at higher level BLE helper classes on top of the current lower level API where an above average understanding of BLE at a packet level is required. A colleague of mine is working on USB for the x40. Other than that, there isn't currently a specific ToDo list for the 52832 and 52840 familiies, but as @tannewt mentioned feel free to add module-level requests to #526, or file specific x40 issues here in a checklist at the top?

arturo182, 2018-02-01:
I guess my problem is that I'm not that familiar with micropython/circuitpython way of working and miss details about the inner-workings as well (and I don't know what I don't know). I'm working on my own PCB with the nRF52840 module I mentioned above and I guess as I'm implementing that I will notice what information I'm missing and then can get more info from you. Is there any resources you would recommend that would tell me more about the lower levels of circuitpython? Any websites/books?

tannewt, 2018-02-01:
@arturo182 A couple resources are https://circuitpython.readthedocs.io/en/latest/docs/design_guide.html and https://circuitpython.readthedocs.io/en/latest/docs/common_hal.html . Other than those, I'd encourage you to join our [Discord chat](https://adafru.it/discord). I'm usually on there and would be happy to video chat with you to get you going.

microbuilder, 2018-02-06:
Closing this issue with the merge of #573, but for follow up see #585, #586, and #587. Issues with the port itself can be handled in a new thread, such as improving `ampy` support for stability, fixing support over the USB CDC interface if possible, etc.

closed 2018-02-06 by microbuilder