adafruit/circuitpython #565: Update py descriptor gen to have HID

ladyada, 2018-01-30:
currently has CDC + MSC only
once key/mouse HID support is added we can extend to do multimedia (and maybe joystick!)

tannewt, 2018-01-31:
What priority do you have for this @ladyada? I don't feel like its required for 3.0 Beta so I'd prefer to wait a bit. Ideally, I'd only assign it to myself once I'm working on it in case someone else beats me to it.

ladyada, 2018-01-31:
Having the busio & CDC+MSC being done and stable is more important than USB HID, so how about after that feels solid - up to y'all whether it's in the beta or not!

tannewt, 2018-02-04:
Ok, I'll unassign for now in case someone wants to beat me to it.

tannewt, 2018-02-09:
Here is where I left on this. Not sure where it was at: https://github.com/tannewt/usb_descriptor/tree/hid

tannewt, 2018-04-05:
This is done in https://github.com/adafruit/usb_descriptor/commit/2edd94c50b67ac2acc665a5b06e34c2625a8f622. Thanks @dhalbert!

closed 2018-04-05 by tannewt