adafruit/circuitpython #568: gamepad.GamePad crashes board

ctgreybeard, 2018-02-01:
I have a CircuitPlayground Express. The lib is 20180128. When I enter the following code the board crashes into safe mode:

Adafruit CircuitPython 2.2.1 on 2018-01-23; Adafruit CircuitPlayground Express with samd21g18
>>> import board
>>> import gamepad
>>> pad = gamepad.GamePad(board.D4)

As I am new to CircuitPython please be gentle ;}

ctgreybeard, 2018-02-01:
Ah, nevermind I think. I just realized that I need to be passing in a DigitalIO object ... sorry ...

dhalbert, 2018-02-01:
But, it should not crash, so we'll leave this open.

ctgreybeard, 2018-02-01:
Hmmm, does't seem to be working ...

import board
import time
import digitalio
import gamepad
pad = gamepad.GamePad(digitalio.DigitalInOut(board.D5))
while True:
    print("pad=", pad.get_pressed())
    time.sleep(0.5)

Doesn't show anything other than 0. And the example code under the gamepad module shows similar results. Should this be a separate issue?

tannewt, 2018-02-01:
@ctgreybeard This issue is good for now. In the meantime, you likely don't need to use GamePad for what you are doing. I'd suggest using DigitalInOut to start with and only use `gamepad` if you find you are missing button presses.

deshipu, 2018-02-01:
How is the button connected to D5?

ctgreybeard, 2018-02-01:
D5 is a button on the Circuit Playground Express. D4 and D5 are buttons on the board, also called BUTTON_A and BUTTON_B.

And I was missing presses which is why I thought to use gamepad. But I think I will use Touch  instead because I can remote that from the board.

tannewt, 2018-02-01:
@ctgreybeard How were you initializing the built-in buttons with DigitalInOut? They require internal pull downs to function properly.

Touch is awesome as well. let us know how it goes on the [Forum](https://forums.adafruit.com/viewforum.php?f=60) or [Discord](https://adafru.it/discord).

deshipu, 2018-02-01:
@tannewt Ah, that would explain it — the gamepad library assumes the buttons are connected to GND and sets pull-ups. Perhaps I should add an option for this somehow.

I need to add the code checking if it's DigitalInOut anyways.

tannewt, 2018-04-02:
Fixed by #732

closed 2018-04-02 by tannewt